zephyrproject-rtos/zephyr #6389: OpenThread: otPlatRandomGetTrue() implementation is not up to spec, may lead to security issues

pfalcon, 2018-02-26:
https://openthread.io/reference/group/plat-random#otplatrandomgettrue specifies:

>otPlatRandomGetTrue
>
>Get true random value sequence.
>
>This function MUST be implemented using a true random number generator (TRNG).

However, implementation in Zephyr:
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/openthread/platform/random.c#L19 uses sys_rand32_get(), which may return pseudo-random value. For TRNG, Zephyr offers entropy subsystem (entropy_get_entropy() function).

pfalcon, 2018-02-26:
This goes with the same disclaimer as the text in parens in https://github.com/zephyrproject-rtos/zephyr/issues/6388#issuecomment-368585206 .

d3zd3z, 2018-05-22:
Also, if the Thread spec does require a TRNG, we should probably document that a device will only be able to be compliant with Thread if the hardware contains a TRNG. It is not possible to implement a TRNG in software (that's kind of the point of the 'True'). It is a little unusual of a requirement, though, since most things will specify a CSPRNG (cryptographically secure pseudo random number generator), even if that is seeded from a TRNG.

nashif, 2018-05-29:
@d3zd3z Can you guys in the SecWG propose a solution and submit it?

d3zd3z, 2018-10-17:
NIST SP 800-90C describes "true", and it should be doable with an entropy source.

agross-linaro, 2018-12-12:
We are going to shelve this for now.  Setting the priority approriately.

d3zd3z, 2019-10-23:
@dleach02 I think this can also be closed as part of your randomness work.

dleach02, 2019-10-24:
@d3zd3z , yep. My intent was to get the csrand stuff merged then make a pass to start putting into use.

 #17736

rlubos, 2019-10-24:
From the OpenThread point of view, this should be resolved in https://github.com/zephyrproject-rtos/zephyr/pull/20106, where I update the OpenThread version used with Zephyr.

The new OpenThread got rid of the `random.c` platform module, and introduced `entropy.c` which is supposed to serve as an entropy source for OpenThread (or rather mbedTLS which OT uses underneath). I've used the Zephyr's entropy subsystem in the new implementation.

d3zd3z, 2019-11-04:
I'm ok closing this, then.

rlubos, 2020-01-17:
No longer an issue with #20106, closing.

closed 2020-01-17 by rlubos