zephyrproject-rtos/zephyr #6392: Trusted Execution: guarantee that Zephyr builds for non-secure images cannot cause Security Exceptions

ioannisg, 2018-02-27:
And implement needed changes for ARM Cortex-M architecture

ioannisg, 2018-02-28:
The problem in detail:
---

A Non-Secure Zephyr image may not be _aware_ of a co-existing Secure image, that runs in parallel.

Therefore, we need to provide a generic way that such an image will **never** access MCU resources that are meant for exclusive use by the Secure image, namely:
* non-banked CPU registers available only to Secure state
* memory regions and peripherals available only to Secure state 
      - memory (including calling of functions that are defined in the Secure region)
      - peripherals

ioannisg, 2018-05-09:
CCing members to give visibility and request feedback: 
@ruuddw , @agross-linaro , @andrewboie , @carlescufi @pizi-nordic

pizi-nordic, 2018-05-09:
Hi.

Basing on your description @ioannisg I see 3 modes of operation on CPU which implements TEE:
- SECURE: TEE is used, image is intended to run in secure domain.
- NON-SECURE: TEE is used, image is intended to run in non-secure domain.
- COMPATIBLE: TEE is not used, image is run on secure domain in order to have full access to hardware.

I think that the importance of the security model setting justify us to force the user to stop for a while and do the selection. So when CONFIG_CPU_HAS_TEE is defined, user should be forced to choose one of CONFIG_<ARCH>_SECURE_FIRMWARE, CONFIG_<ARCH>_NONSECURE_FIRMWARE and CONFIG_<ARCH>_COMPATIBLE_FIRMWARE (naming is not perfect, but this is just an example).

Next, if we mark the compatible mode as default one, then backward compatibility with SoC lacking TEE is maintained and users which are not going to use TEE just keep the default, which is the easiest to use.

ioannisg, 2018-05-16:
@pizi-nordic 
>COMPATIBLE: TEE is not used, image is run on secure domain in order to have full access to hardware.

This is, rather, the case when the user either ignores the TEE entirely, or aims at having full access to MCU resources (as you say). The first ensures backwards compatibility (for example, for software that migrates from ARMv7-M/ARMv6-M to ARMv8-M platform with Security Extensions).

> user should be forced to choose one of CONFIG_<ARCH>_SECURE_FIRMWARE, CONFIG_<ARCH>_NONSECURE_FIRMWARE and CONFIG_<ARCH>_COMPATIBLE_FIRMWARE

In principle, I am fine with having a default mode (COMPATIBLE). This makes the developer more conscious of what is going on. However, this will, rather, be an inactive option, right? I mean, I do not see at this point, where an #ifdef CONFIG_ARM_COMPATIBLE_FIRMWARE could be used. So, it is equivalent to not having a default option at all.

pizi-nordic, 2018-05-22:
@ioannisg 

You are right, that the CONFIG_COMPATIBLE_FIRMWARE will likely never appear in the code. The presence of this option in the configuration only makes the user aware of his choice.

ruuddw, 2018-05-25:
Here's my first feedback on these ideas and proposals.

"A Firmware image that boots the MCU is by default running in Secure state. [This is true for ARMv8-M and ARC (@ruuddw ).]" -> Correct, ARC always boots in Secure Kernel Mode

I agree with the general approach and use-cases. Current ARC implementation for SecureShield on EMSK (em7d configuration) follows the 'compatible' approach: runs everything in Secure mode. I'm not really in favor of the term 'compatible' by the way.
I fully support the notion of being able to build a 'secure' and a 'normal' image. But I'd rather separate a few steps/concepts that seem to be combined here: the 'how to build' (normal or secure firmware image) and the 'what to build' (zephyr RTOS & applications, or only some 'baremetal' Secure bootloader and Secure services). I think there are 3 parts:
1:  build a Normal or Secure firmware image : set correct compiler/linker options, linker command file etc.
2:  what should go into the image: Zephyr kernel (+application), bootloader, 'Secure' services that can be used from 'Normal', ...
3:  optional: combine the normal and secure images into an image that can be deployed to a target, and runner scripts etc

ioannisg, 2018-05-28:
@ruuddw thanks for your feedback. Here are some responses to your comments.

>  I'm not really in favor of the term 'compatible' by the way.

That's totally OK; we could discuss the exact terminology under a separate issue.

> But I'd rather separate a few steps/concepts that seem to be combined here

So, in this issue, my intention was, mainly, to address the _"how-to-distinguish"_ problem, from the perspective of the product maker/developer. Hence, the proposed **top-level** `arch` build-directives, `CONFIG_SECURE_FW`, `CONFIG_NONSECURE_FW`.

I argued that in systems with TEE we have 3 different Firmware categories:
- Secure-aware
- Non-Secure-aware
- TEE-unaware (boils down to Secure-by-default, but not using TEE)

I believe, that is actually the first step: decide what kind of firmware you want to build.
**From the ARC-perspective; do you think there is any other possible kind of Firmware (that does not fall into the above categories)?**
Same questions to @agross-linaro, regarding ARM.

> 'how to build' (normal or secure firmware image) and the 'what to build' (zephyr RTOS & applications, or only some 'baremetal' Secure bootloader and Secure services). 

Correct, these are the following things we need to address and design.

> build a Normal or Secure firmware image : set correct compiler/linker options, linker command file etc.

Exactly. To some extend only, this is addressed (for ARM) in #6748 but we need to decide on a generic, "all-archs" approach for it.

>  what should go into the image: Zephyr kernel (+application), bootloader, 'Secure' services that can be used from 'Normal', ...

Is this something, more or less, determined by prj.conf of the user application? [E.g. to skip the RTOS kernel; isn't it sufficient to have CONFIG_MULTITHREADING=n?]

> optional: combine the normal and secure images into an image that can be deployed to a target, and runner scripts etc

Agree. This is the last step, once all sub-images are built.

ioannisg, 2018-10-19:
Fully supported in PR #10544

ioannisg, 2018-10-24:
Closing this since #10544 has been merged.

closed 2018-10-24 by ioannisg