zephyrproject-rtos/zephyr #6399: How to using the PPI chanels from 20-31 in Nrf5 chip?

niuton, 2018-02-28:
I have seen the Nrf52832 spec and can't know exactly what I have to do to using  programmed channels. Please help me

nashif, 2018-08-14:
@cvinayak ping, can you answer please

cvinayak, 2018-08-14:
@niuton these are pre-programmed PPI channels as stated in the product specification. These are very specifically decided in SoC to interconnect radio with other peripherals, currently these are not used in Zephyr as the native BLE controller uses the Radio uses programmable PPIs in its design/implementation. In the future, if some of these can be used by the BLE controller, then some programmable PPI will be made available for the applications.

Its best you do not use it alongside Bluetooth in Zephyr applications.

closed 2018-08-14 by cvinayak