zephyrproject-rtos/zephyr #6400: samples/net/http_client: Failed to connect to samples/net/http_server

tarunkum, 2018-02-28:
I was trying to test http_server along with http_client with two different frdm_k64f boards.
 The steps I have followed for http_server & http_client is as below:

cd zephyr/samples/net/http_client
 mkdir build && cd build && cmake -GNinja -DBOARD=frdm_k64f ../
 ninja
 cp -vr outdir/frdm_k64f/zephyr.bin /run/media/MBED/

cd zephyr/samples/net/http_server
 mkdir build && cd build && cmake -GNinja -DBOARD=frdm_k64f ../
 ninja
 cp -vr outdir/frdm_k64f/zephyr.bin /run/media/MBED1/

•Here I am trying make one frdm_k64f as http_client & other one as http_server.

•I have done below changes in prj.conf files in http_client sample apps.

	CONFIG_NET_APP_MY_IPV6_ADDR="2001:db8::1"
	CONFIG_NET_APP_MY_IPV4_ADDR="192.0.2.2"
	CONFIG_NET_APP_PEER_IPV6_ADDR="2001:db8::2"
	CONFIG_NET_APP_PEER_IPV4_ADDR="192.0.2.1"

	On http_sever sample I have done any changes in prj.conf file.

•After flashing both codes http_client is not able to connect to http_server on console.

On http_client console :-

***** BOOTING ZEPHYR OS v1.10.99 - BUILD: Feb 28 2018 08:56:05 *****
Zephyr Shell, Zephyr version: 1.10.99
Type 'help' for a list of available commands
shell> [dev/eth_mcux] [INF] eth_mcux_phy_event: Enabledduplex mode.
[http-client] [INF] main: --------Sending 20 sync request--------
[http-client] [INF] do_sync_http_req: [1] Send /index.html
[http/client] [DBG] http_client_send_req: (0x20003838): Cannot connect to server (-60)
[http-client] [ERR] do_sync_http_req: Cannot send GET request (-60)
[http/client] [DBG] http_closed: (0x20003838): [0x20000000] connection closed
[http-client] [INF] main: Done!

I have tested on latest commit today

rveerama1, 2018-02-28:
@tarunkum Can you try with https://github.com/zephyrproject-rtos/zephyr/pull/6167 PR.

nashif, 2018-03-27:
@tarunkum ping (FYI @vikramx2112)

tarunkum, 2018-03-28:
@rveerama1 

 I took the latest branch use the same configuration as I mentioned earlier. I am not able to find the communication between server-client. 

Please find the below message log of http_client  on serial window.

***** BOOTING ZEPHYR OS v1.11.99 - BUILD: Mar 28 2018 06:27:55 *****
[dev/eth_mcux] [INF] eth_mcux_phy_event: Enabled 100M full-duplex mode.
~x~] [INF] main: --------Sending 20 sync request--------
[http-client] [INF] do_sync_http_req: [1] Send /index.html
[net/tcp] [DBG] net_tcp_change_staa8): [0x200 CLOSED (0) (2)
[net/tcp] [DBG] net_tcp_trace: (0x200038a8): [0x200029f8] pkt 0x200055dc s0(3 (0/0) flags uaprSf win 28b
[http/client] [DBG] http_client_send_req: (0x200038a8): Cannot connect to server (-60)
[http-client] [ERR] do_sync_http_req: Cannot send GET request (-60)
[http/client] [DBG] http_closed: (0x200038a8): [0x20000000] connection closed
[net/tcp] [DBG] net_tcp_change_state: (0x200038a8): [0x200029f8] state SYN_SENT (2) => CLOSED (0)
[net/tcp] [DBG] net_tcp_release: (0x200038a8): [0x200029f8] Disposed of TCP connection state
[http-client] [INF] main: Done!

rveerama1, 2018-04-27:
@tarunkum http_server and http_client samples are for different purpose. They are implemented in way to communicate each other. http_server sample expects different client and http_client expects different server capabilities.

The reason in your test these samples can not even establish TCP connection is http_server listening on port 80 and client tries to connect on port 8000. You have changed these configuration
CONFIG_NET_APP_MY_IPV6_ADDR="2001:db8::1"
CONFIG_NET_APP_MY_IPV4_ADDR="192.0.2.2"
CONFIG_NET_APP_PEER_IPV6_ADDR="2001:db8::2"
CONFIG_NET_APP_PEER_IPV4_ADDR="192.0.2.1"

but forgot to modify port numbers.
--------------------------------------------------------------------
diff --git a/samples/net/http_client/src/config.h b/samples/net/http_client/src/config.h
index ae5cff887..d43561fd3 100644
--- a/samples/net/http_client/src/config.h
+++ b/samples/net/http_client/src/config.h
@@ -17,7 +17,7 @@
 #if defined(CONFIG_HTTPS)
 #define SERVER_PORT  4443
 #else
-#define SERVER_PORT  8000
+#define SERVER_PORT  80
 #endif
---------------------------------------------------------------------------------------

I have tried modifies port number and they are able to created at-least TCP connection. But you do get
[http-client] [ERR] do_sync_http_req: Cannot send GET request (-60).

Reason I mentioned already is both the samples expects different counterparts as client and server.
So this issue is not valid, hence closing.

closed 2018-04-27 by rveerama1