zephyrproject-rtos/zephyr #6413: net_mgmt.h API can't be used as documented

mbolivar, 2018-02-28:
The net_mgmt.h and net_event.h APIs have issues which mean they can't be used correctly:

- Thread-safety (not so serious)
- Definition of "event masks" contradicts fact that the values are *not* independent bit masks that can be logically ORed together (seems quite serious)

*Thread safety*:

Public API functions like `net_mgmt_add_event_callback()` and `net_mgmt_del_event_callback()` use thread-unsafe sys_slist_t primitives directly; internal API functions like `mgmt_add_event_mask()` also do thread-unsafe operations. This is a problem since user-facing routines like `net_app_init()` add callbacks; these could race with other user/system code.

*"Event masks"*:

Public API functions like `net_mgmt_init_event_callback()` say they take "A mask of relevant events for the handler", and code such as https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/app/init.c#L222 does OR values together as if the bits in each `NET_EVENT_xxx` macro were different.

But the macros used to define them show that each `NET_EVENT_xxx` actually is made of multi-bit fields:

https://github.com/zephyrproject-rtos/zephyr/blob/master/include/net/net_mgmt.h
https://github.com/zephyrproject-rtos/zephyr/blob/master/include/net/net_event.h

The values in these fields overlap, which means they're not masks; each event is a separate value that cannot be combined with others.

For an example of why this is wrong, note that `NET_EVENT_IPV6_MADDR_DEL | NET_EVENT_IPV6_ADDR_ADD` is equal to the value for `NET_EVENT_IPV6_PREFIX_ADD`.

The handling code for this is also broken. Note that mgmt_run_callbacks() just tests to see if *any* layer, layer code, and command bits overlap between a callback's event_mask and the one in the `struct mgmt_event_entry` before running the callback handler:

https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/ip/net_mgmt.c#L139

That's also wrong since these fields can share bits without referring to the same events. For example, consider what happens if `mgmt_event->event == NET_EVENT_IPV6_MADDR_ADD` and `cb->event_mask == NET_EVENT_IPV6_ADDR_ADD`. This could even be an issue between layers: for example, note that `NET_REQUEST_BT_ADVERTISE` and `NET_EVENT_IPV6_MCAST_JOIN` also pass this test.

In general, this API seems to be broken and should be fixed.

mbolivar, 2018-02-28:
@mike-scott FYI

tbursztyka, 2018-03-01:
Well you missed that discussion back then, but this event/request/masq was all made on purpose: to save space. It's ok to get events the callback is not waiting for, and thus do nothing: the callback gets the right mgmt_request, so up to the callback to verify it, rather than bloating the net_mgmt_event_callback (and net_mgmt.c internals etc...).

Now if NET_REQUEST_BT_ADVERTISE and NET_EVENT_IPV6_MCAST_JOIN pass, then yes, there is a bug in net_mgmt.c line 149 and below.

the masq is not supposed to match more than 1 layer, same for the code, so there is definitely a bug there : should be '==' not '&'

mbolivar, 2018-03-01:
Hm, so you're saying that callback handlers *must* check the actual event code, right?

In other words, there are really two data types here:

- actual event codes, which have interesting bitfields inside
- event *sets*, which are ORed together event codes

Event sets act like Bloom filters:

- false negatives are impossible: `(event_code & event_set) != event_code` means the code is not in the set
- false positives are possible: `event_code & event_set == event_code` doesn't guarantee the code is in the set

With that and your bug fix PR (https://github.com/zephyrproject-rtos/zephyr/pull/6418) I think I see the intention now, but still think that issues remain with the API, and that it is hard to use / could use better docs.

**Remaining issues**

The fact that false positives are possible means some APIs are still racy. Here is an example:

1. User calls net_mgmt_event_wait(NET_EVENT_1 | NET_EVENT_2, ....)
2. NET_EVENT_3, which matches the `NET_EVENT_1 | ... _2` event set, occurs and is handled
3. net_mgmt_event_wait() returns to user with `*raised_event == NET_EVENT_3`
4. User thread is scheduled out, then NET_EVENT_1 happens, and is completely handled by the system
5. User thread scheduled in, notices raised_event didn't match, and starts waiting for the desired events again

In general I believe this means that net_mgmt_event_wait and net_mgmt_event_wait_on_iface cannot be used safely. Perhaps they should just be restricted to wait on exactly one event code (not an event set), as conveniences so users don't have to initialize their own semaphores etc. for simple situations.

**Example misuses / misleading documentation**

These users don't check the actual codes:

https://github.com/zephyrproject-rtos/zephyr/blob/master/tests/net/dhcpv4/src/main.c#L523
https://github.com/zephyrproject-rtos/zephyr/blob/master/tests/net/mld/src/main.c#L154
https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/net/coap_client/src/coap-client.c#L165
https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/net/telnet/src/telnet.c#L25

None of the documentation seems to make the requirement for checking the code clear:

- Doxygen documentation for net_mgmt.h, net_event.h
- Detailed docs in http://docs.zephyrproject.org/subsystems/networking/network-management-api.html
- The referenced example, i.e. https://github.com/zephyrproject-rtos/zephyr/blob/master/tests/net/mgmt/src/mgmt.c#L124, doesn't actually check the code (other asserts elsewhere in the file do)

(I've sent a PR to fix these example problems, https://github.com/zephyrproject-rtos/zephyr/pull/6421)

mbolivar, 2018-03-02:
> Yes looks like you debugged what were my intentions and what I put in the doc, not to mention about the actual code bugs. Thanks for the PR that clarifies things well

What do you think about the remaining issues though?

tbursztyka, 2018-03-02:
The sync call net_mgmt_event_wait() behaves the same as with a callback: up to it to check the request.

For the missing event, it's totally fine: net_mgmt is not meant to be reliable. Actually, if the system has too many notifications compared to the size of the internal ring, new events will overwrite the earliest, etc... Loosing event on emitter or receiver side is fine.

mbolivar, 2018-03-02:
> For the missing event, it's totally fine: net_mgmt is not meant to be reliable. Actually, if the system has too many notifications compared to the size of the internal ring, new events will overwrite the earliest, etc... Loosing event on emitter or receiver side is fine.

OK, that's true, but I'm not sure this is "fine", what makes you say so?

Some events are really critical to certain use cases (like link up / down); if net_mgmt is just for logging purposes or so then should we discuss an additional API for reliable event delivery with caller-allocated resources instead of a global queue?

mbolivar, 2018-03-02:
Especially since the queue size is controllable by Kconfig already, and the "set" of events which are handled is controllable by the user, the fact that sets have false positives (due to this choice of data structure) contributes directly to the possibility of lost events, since events nobody cares about within layers people do care about can be added to the queue.

tbursztyka, 2018-03-02:
Well, about the sync method: there is no code using it, perhaps we may remove it in the future as it generates a potential loss from the receiver side. Or maybe we could mandate something: it should be used so wait for 1 command only, and the net_mgmt core would specifically check for an exact match only in this case. What do you think?

But on the emitter side, ring size should be tweaked to lower the chance of loss. That needs to be set case by case really. Such loss could occur only on net stack high load. It's like netlink in linux: you don't have the garanty to catch all the event in case of high load.

carlescufi, 2018-03-02:
Sorry @mbolivar did not realize this was going to close the issue, I replaced "Fixes" with "Addresses .."

mbolivar, 2018-03-02:
> Sorry @mbolivar did not realize this was going to close the issue, I replaced "Fixes" with "Addresses .."

No problem -- I've updated the issue description as you requested to reflect what's happened here.

tbursztyka, 2018-03-06:
So, isn't this solved already then?

mbolivar, 2018-03-21:
> Well, about the sync method: there is no code using it, perhaps we may remove it in the future as it generates a potential loss from the receiver side. Or maybe we could mandate something: it should [...]

This is roughly what I had proposed earlier but with changes so it could work; no objections as it fixes the race within the current constraints.

> 
> But on the emitter side, ring size should be tweaked to lower the chance of loss. That needs to be set case by case really. Such loss could occur only on net stack high load. It's like netlink in linux: you don't have the garanty to catch all the event in case of high load.

The inability to fully control which events get queued feels like the wrong tradeoff to me and others I've discussed with, in addition to the API being confusing to use.

I would rather spend a bit of extra memory per event to guarantee that only events I care about are being queued, so I can manage the queue size according to my own needs without worrying about new event codes being added upstream messing me up between releases, due to false positives.

You're the maintainer, though.

Potentially there's a band-aid here that might make us both happy. Here is a summary of the situation today:

| "Layer"/area | "Layer" code | Layer | Num. of events |
|--------------|--------------|-------|----------------|
| Core         | 0x001        | L1    | 2              |
| IPv6         | 0x060        | L3    | 16             |
| IPv4         | 0x004        | L3    | 3              |
| BT           | 0x155        | L2    | 4              |
| 802.15.4     | 0x154        | L2    | 19             |

In all cases except 15.4, the number of events permits using the 16 bits of "command code" as a true bitmask, preventing the possibility of false positives, at least within a layer.

How about making that change?

tbursztyka, 2018-03-22:
@mbolivar btw, fixed an issue with event notification ordering (it was obvious when I implemented wifi scan: last result was always coming after "scan done" event, though they were inserted in a reverse order) see #6743 

Currently the number of events is what you listed, but nothing prevents this list to grow again, making let's say ipv6 stuff not being a mask also etc... 

I can make mgmt_is_event_handled() a bit more clever (matching exactly layer and code).

mbolivar, 2018-03-22:
> Currently the number of events is what you listed, but nothing prevents this list to grow again, making let's say ipv6 stuff not being a mask also etc...

Even with >16 events and only checking if any bits overlap between command code and global event mask (as in PR #6743), I still think it's worth avoiding possible false positives -- and thus memory pressure on the queue -- when we can. Wouldn't you agree?

This can be done by acting like a mask as long as possible and minimizing overlap between command codes afterwards, like this:

```
code 1 = BIT(0),
code 2 = BIT(1),
...
code 16 = BIT(15),
code 17 = BIT(0) | BIT(1),
code 18 = BIT(2) | BIT(3),
...
code 24 = BIT(14) | BIT(15),
etc.
```

The difference in the number of false positives gets bigger the more events there are:

- 17 events: consecutive integer command codes 89 false positive pairs, "bits" 2 false positive pairs (see https://gist.github.com/mbolivar/b292cc1e012fddc85e12caed85b477a0)
- 18 events: integers 99, "bits" 4
- 19 events: integers 114, "bits" 6
- 20 events: integers 126, "bits" 8
- 21 events: integers 143, "bits" 10
- 22 events: integers 161, "bits" 12
- 23 events: integers 182, "bits" 14

> I can make mgmt_is_event_handled() a bit more clever (matching exactly layer and code).
> [edit] made into PR #6743

Thanks, reviewed!

mbolivar, 2018-03-22:
> I still think it's worth avoiding possible false positives -- and thus memory pressure on the queue -- when we can.

Noting that I'd still prefer to avoid it altogether :), but even within the current implementation, there's easy room for improvement.

laperie, 2018-04-12:
clearing 1.12 milestone as there is no clear action plan for the issues described; we will know the version milestone once the discussion is over.

jukkar, 2018-05-02:
It seems that #6754 is not going to be merged so do we need this bug? @mbolivar what do you think?

mbolivar, 2018-05-09:
> It seems that #6754 is not going to be merged so do we need this bug? @mbolivar what do you think?

Do you want to keep it at low priority? I'm OK with closing for now if not.

jukkar, 2018-05-11:
Lets close this one at this point and re-open if needed.

commit 7255d7bb (2018-03-02): net/mgmt: Fix how event_mask is handled on layer and layer code part

Exact match is the rule here.

Fixes #6413
commit f835f7f6 (2018-03-02): net/mgmt: Protect the event_callback list when manipulating it.

Adding a lock where relevant.

Fixes #6413
closed 2018-05-11 by jukkar