zephyrproject-rtos/zephyr #6439: ESP32 doesn't compile in master

locomuco, 2018-03-03:
used toolchain version:
```
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
```

compilation error in samples/hello_world:
```
...
[ 30%] Building C object zephyr/CMakeFiles/zephyr.dir/arch/xtensa/soc/esp32/esp32-mp.c.obj
[ 31%] Building C object zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/cpu_idle.c.obj
[ 33%] Building C object zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/fatal.c.obj
[ 34%] Building ASM object zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/xtensa-asm2-util.S.obj
/home/matthias/workspace/zephyr/arch/xtensa/core/xtensa-asm2-util.S: Assembler messages:
/home/matthias/workspace/zephyr/arch/xtensa/core/xtensa-asm2-util.S:161: Error: unknown opcode or format name 'wsr.lbeg'
/home/matthias/workspace/zephyr/arch/xtensa/core/xtensa-asm2-util.S:163: Error: unknown opcode or format name 'wsr.lend'
/home/matthias/workspace/zephyr/arch/xtensa/core/xtensa-asm2-util.S:165: Error: unknown opcode or format name 'wsr.lcount'
/home/matthias/workspace/zephyr/arch/xtensa/core/xtensa-asm2-util.S:198: Error: unknown opcode or format name 'rsr.lbeg'
/home/matthias/workspace/zephyr/arch/xtensa/core/xtensa-asm2-util.S:198: Error: unknown opcode or format name 'rsr.lend'
/home/matthias/workspace/zephyr/arch/xtensa/core/xtensa-asm2-util.S:198: Error: unknown opcode or format name 'rsr.lcount'
zephyr/CMakeFiles/zephyr.dir/build.make:198: recipe for target 'zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/xtensa-asm2-util.S.obj' failed
make[2]: *** [zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/xtensa-asm2-util.S.obj] Error 1
CMakeFiles/Makefile2:521: recipe for target 'zephyr/CMakeFiles/zephyr.dir/all' failed
make[1]: *** [zephyr/CMakeFiles/zephyr.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

```

locomuco, 2018-03-03:
@lpereira FYI

lgl88911, 2018-03-05:
I meeting same issue， My toolchain gcc vesrion is 6.3.0

lpereira, 2018-03-05:
This seems to be related to the new assembly framework.  @andyross, could you take a look here?

In the meantime, please use an [older version of the SDK](https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz).

@lgl88911 What toolchain are you using that comes with GCC 6.2.0?

andyross, 2018-03-05:
I can't reproduce this, unfortunately.  I just pulled and installed a clean version of the "g6c4433a" toolchain (md5sum 3e00f8faa7360ebfc39971d2d2f3522d xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz, just in case they're playing games) and got a working build and executable from a bunch of different tests.

Can you verify that you're using the toolchain you think you are?  ESPRESSIF_TOOLCHAIN_PATH is set to the new one and not the old one?  Try make with VERBOSE=1 and manually check the compiler path?

@lgl88911 : you're most definitely using the wrong toolchain.  That one is for the qemu emulator.  Xtensa toolchains are not portable across different variants.  You need one built for the particular CPU in question.

lpereira, 2018-03-05:
@locomuco A shot in the dark here: what's the esp-idf version you're using? This code in particular depends on something that `core-isa.h` defines, and this file is provided in esp-idf.

lpereira, 2018-03-05:
Do you have the full build output? I think I've nailed it down here to a set (but empty) ZEPHYR_GCC_VARIANT environment variable that was superseding ZEPHYR_TOOLCHAIN_VARIANT. This was calling the wrong assembler. By fixing the environment variables and a clean build directory, I got this CMake output:

```
-- Generating zephyr/include/generated/generated_dts_board.h
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /.../esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
```

Which shows that it correctly found the ESP32 toolchain.  The build works as usual.  I'm sending just a simple PR that enables Zephyr to be built with the latest ESP-IDF.

closed 2018-03-06 by nashif

locomuco, 2018-03-06:
@lpereira thanks, the variables fixed it on my side too

lgl88911, 2018-03-10:
@lpereira Thanks, I export ZEPHYR_GCC_VARIANT=zephyr cause"ZEPHYR_GCC_VARIANT is deprecated, please use ZEPHYR_TOOLCHAIN_VARIANT" and using the wrong toolchain. Now remove ZEPHYR_GCC_VARIANT=zephyr ,compile pass

rolandma, 2019-11-22:
Hello, I'm using the zephyr-sdk-0.10.3 and I can't build the "hello-world" with esp32 board. The error is the same as the original posted.
Thanks.

PR #6467 (2018-03-06): soc: esp32: Enable building with newer ESP-IDF
ESP-IDF commit cb31222e added the dependency on a file named
"sdkconfig.h", which is equivalent to "autoconf.h" generated by kbuild
used in Zephyr.  It does not depend on anything from that file, though,
so just provide an empty file to keep the compiler from complaining.

Fixes #6439