zephyrproject-rtos/zephyr #6450: Several devices of same type on same bus - how to address?

webhive, 2018-03-04:
Hello! I am trying to implement device driver, but researching samples found what drivers model assumed we have single device of any type. For example HTS221 on I2C bus. What if I need to add several devices on same bus. We have only single HTS221_GPIO_DEV_NAME. 

Reviewed several device drivers Kfiles - most have device name and device address. Looks like no way now to assign several addresses to same bus. 

May be I am wrong and dont understand something. Anyway any help could be good.

MaureenHelm, 2018-03-08:
Most or all of the sensor drivers assume only one instance, but could be refactored to support multiple instances. The uart and i2c drivers are better about supporting multiple instances, since it is much more common to need multiple uarts than it is to need multiple humidity sensors.

closed 2018-03-28 by MaureenHelm

lodup29, 2021-06-30:
@MaureenHelm I believe that the I2C driver model also suffers from the same issue as reported by @webhive. Yet, more than one instance of the same driver seems like a very comme use case on a given I2C bus.

MaureenHelm, 2021-07-15:
> @MaureenHelm I believe that the I2C driver model also suffers from the same issue as reported by @webhive. Yet, more than one instance of the same driver seems like a very comme use case on a given I2C bus.

Is there a specific I2C driver you're looking at? We refactored a lot of drivers to support multiple instances last year, and have been enforcing it on new drivers since then.

https://docs.zephyrproject.org/latest/guides/dts/howtos.html#write-device-drivers-using-devicetree-apis

symstopje, 2023-11-22:
@MaureenHelm does  tmp116 driver support multiple instances? how to check if driver supports multiple?