zephyrproject-rtos/zephyr #6452: Jumbled Console log over USB

ajaykish, 2018-03-05:
Tested with sample/subsys/usb/console (zephyr 1.11.rc1 release) on Arduino 101 connected to Linux host:

Plug the board into a Linux host. 
The board will be detected as a CDC_ACM serial device. To see the console output
from the zephyr board, used  command  "minicom -D /dev/ttyACM0".

The console output we are getting is jumbled:
ld!x8
Hello World! x86
HellWorld! x86
              llo Word! x6
                          Hello Wordx6
                                      Hello World! x86
Hello World!x6
              Hello World! x86
Hel Wol! x6
HelWrdx6
        Hlo Worl!x6
                   Hello o6
                           Heo rl!x6
                                    Hello World! x86
Hello World! x86
Hello World! x86

Issue is easily reproducible in almost every iteration.

finikorg, 2018-03-15:
This looks related to  #3614

finikorg, 2018-05-30:
@nashif This is not fixed yet

jfischer-phytec-iot, 2018-05-30:
Can not confirm, tested with current master, connected to xhci and ehci.

finikorg, 2018-06-04:
@jfischer-phytec-iot Do you mean it works for you? I just tested on master + PR #4661 - I see the same issue described in this bug and bug #3614.

jfischer-phytec-iot, 2018-06-04:
@finikorg it works for me, also I noticed that the errors with Arduino 101 are host controller dependent.

galak, 2018-12-06:
@finikorg is still an issue?

laperie, 2019-03-11:
That's a true low-priority bug, we don't see it on any other hardware than Arduino 101

laperie, 2019-03-13:
Just like in #5678, we're closing it based on Arduino 101 not being actively maintained

closed 2019-03-13 by laperie