zephyrproject-rtos/zephyr #6455: k_sleep() fails on ESP32 sometimes

nniranjhana, 2018-03-05:
used toolchain version:
`xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-61-gab8375a) 5.2.0`
I had added a loop that runs `k_sleep(1000)` in the hello world sample (while narrowing down the likely issue from another test case). I had added print statements before and after, and as per the log below, it returns from sleep the first time, but fails the second time:
`
***** BOOTING ZEPHYR OS v1.10.99 - BUILD: Mar  5 2018 19:12:04 *****
Hello World! xtensa
==> going to sleep      ==> woke up
==> going to sleep
`

nniranjhana, 2018-03-05:
@lpereira @youvedeep FYI

lpereira, 2018-03-05:
This is most likely related to #6346

lpereira, 2018-03-05:
In fact, it looks like the same issue. Closing this one as @andyross is already working on the matter.

PR #6465 (2018-03-06): xtensa/asm2: Fix stack pointer during preemption spills
When returning into a different thread than we interrupted, we
obviously need to spill all the existing register windows to make sure
all their values are in the old thread's stack.  But the code to do
this forgot to reset the current stack pointer to the value it had at
interrupt time (it was still pointing to the saved context below
that), so the caller of the interrupted function was spilling to the
wrong spot.

This wouldn't show up as an instant failure, it would only happen when
switching BACK to the improperly-spilled thread.  And  [... cut]
closed 2018-03-05 by lpereira