zephyrproject-rtos/zephyr #6470: tests/crypto/ecc_dh results in FATAL EXCEPTION on esp32

pswarnak, 2018-03-06:
[tests/crypto/ecc_dh/testcase.yaml#crypto.ecc_dh](https://github.com/zephyrproject-rtos/zephyr/blob/v1.11.0-rc1/tests/crypto/ecc_dh/testcase.yaml) results in FATAL EXCEPTION on esp32 board. The test results in random failure. It sometimes just gets hung without any output and crashes with below exception, the other times.

**Configuration:**
Arch - xtensa 
Board - esp32

**Failure log:**
```
***** BOOTING ZEPHYR OS v1.10.99 - BUILD: Mar  6 2018 01:09:55 *****
Running test suite test_ecc_dh_fn
===================================================================
starting test - test_ecc_dh
starting test - Performing ECC-DH tests:
Performing cavp_ecdh test:
Test #1: ECDHNIST-p256
  Vector #00 check Z - success
  Vector #01 check Z - success
  Vector #02 check Z - success
 ** FATAL EXCEPTION
 ** CPU 0 EXCCAUSE 20 PS 0x00060720 PC 0xffffffff VADDR 0x00000000
 **  A0 0x80082d26  SP 0x3ffb4290  A2 0x3ffb42c0  A3 0x00000000
 **  A4 0x3ffb4340  A5 0x00000008  A6 0xdc90a810  A7 0x00000000
 **  A8 0x00000000  A9 0xdc90a810 A10 0x7e3a371b A11 0x00000000
 ** A12 0x00000000 A13 0x00000008 A14 0x00000000 A15 0x3ffb42c0
 ** LBEG 0xe2b6d5a5 LEND 0x2c6dd31d LCOUNT 0x00000000
 ** SAR 0xbfddc738
Current thread ID = 0x3ffb3a34
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
```

lpereira, 2018-03-06:
This works with the previous ASM stuff. @andyross, could you please take a look?

andyross, 2018-03-06:
Will take this one.  It's definitely an issue with asm2 interrupt handling.  Putting an irq_lock() around the test makes it work, and adding a counter to the handler shows that the failure happens stochastically after a few dozen interrupts.   Smells like a problem with context restore, maybe a race between interrupt levels.

Seems to effect esp32 and not qemu_xtensa, which I guess isn't so surprising since timing will be more deterministic in a simulator.  Might have to do with register count (qemu has 32, not 64) affecting the probability.

andyross, 2018-03-06:
Pffft.  Nothing so fancy.  I, uh, had forgotten to spill the shift register in interrupt entry.  Other tests weren't catching that because they didn't shift enough (you have to hit the interrupt between the load of SAR and the use in a shift, and that's generally exactly one cycle).  Crypto does a lot of shifting.  Seems to be fixed with https://github.com/zephyrproject-rtos/zephyr/pull/6492

nashif, 2018-03-06:
this is now fixed.

closed 2018-03-06 by nashif

pswarnak, 2018-03-07:
Thank you @andyross for the quick fix. 
Checked all the 4 test cases(ecc_dsa, aes, sha256, ecc_dh) that were failing earlier, works fine now.