zephyrproject-rtos/zephyr #6514: samples/drivers/i2c_fujitsu_fram: "Error writing to FRAM! error code (-5)"

spoorthik, 2018-03-08:
Sample application at samples/drivers/i2c_fujitsu_fram is failing after writing first byte. 

Log below:
```
Wrote 0xAE to address 0x00.
Error writing to FRAM! error code (-5)
```
Steps to reproduce:
1. cd zephyr.git/samples/drivers/i2c_fujitsu_fram
2. mkdir build/ && cd build/
3. cmake -DBOARD=quark_se_c1000_devboard .. && make flash

spoorthik, 2018-03-26:
With  #6610, the test case is now passing. 

Sample log below: 
```
***** BOOTING ZEPHYR OS v1.11.99 (delayed boot 1000ms) - BUILD: Mar 26 2018 05:18:13 *****
 Wrote 0xAE to address 0x00.
 Wrote 0x86 to address 0x01.
 Read 0xae from address 0x00.
 Read 0x80 from address 0x01.
 Wrote 16 bytes to address 0x00.
 Read 16 bytes from address 0x00.
 Data comparison successful.
```

nniranjhana, 2018-09-07:
This test is failing on arduino_101, x86 core of quark_se_c1000 with the following log:

```
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS v1.13.0-rc3 (delayed boot 1000ms) *****
 Error writing to FRAM! error code (-5)

```

nniranjhana, 2018-09-07:
Letting this issue remain closed, as this seems to a different issue tracked in #9849

closed 2018-09-07 by nniranjhana