zephyrproject-rtos/zephyr #6515: boards: em_starterkit:  the reset mechanism is not stable

vonhust, 2018-03-08:
em starterkit just has a simple hardware reset mechanism as it's just a FPGA protype board. 
This will affect the continuous automatic tests on it. 

It's better to use some way ( opencd cmds or gdb cmds, or hardware) to improve it

ruuddw, 2018-03-08:
This relates to #6267 sanitycheck script support for device testing instead of qemu simulation: ideally boards should be (hardware) reset before running a next test.

inakypg, 2018-03-08:
In our automation HW we realized that the only way to get these things to properly reset is to plug all the cables out, let the board sit for a few secs (for all the capacitance to drain) and then replug.

This applies to most MCUs we've put in there -- seems to work most of the time, but not all//consistently.

You definitely need HW assist for that (http://www.neobits.com/digital_loggers_lpc7_digital_loggers_web_power_p5976274.html?atc=gbp&gclid=EAIaIQobChMIw7Ooobvd2QIVAgppCh0Plwc4EAQYASABEgJC9PD_BwE, https://www.yepkit.com/products/ykush).

We are in the final stages of open sourcing the HW automation solution that supports those things, so we'll be able to share it.

commit c63298ea (2018-06-08): arch: arc: improve the reset code

This commit improves the reset of arc:
* make the processor in the correct status
* clear interrupt related regs

this may improve or fix #6515
closed 2018-06-08 by MaureenHelm