zephyrproject-rtos/zephyr #6522: Should have a "dumb" O(N) scheduler

andyross, 2018-03-08:
Right now our scheduler has to manage an array of 32 separate priority queues, virtually all of which are going to be empty.  Typical applications often never have more than one runnable thread ever.  That's 128 bytes of RAM wasted and a fairly large amount of code to manage this scheduling algorithm.

For smaller apps, or even large ones with small thread counts, it would be nice to have a configurable "dumb" scheduler which simply did a linear walk over the runnable threads to pick the highest priority one.  This would be smaller in both code and RAM footprint, and probably execute faster too.

Fancier integration would be able to select this automatically.  For example if there are only ~3 static threads defined and k_thread_create() (or equivalent) is not referenced at link time, we could fallback automatically to the dumb implementation.

pfalcon, 2018-03-14:
+1. I also remember that some people previously expressed concerns/issues with the current scheduler. I'm not sure if we can talk about "pluggable" schedulers, but have an alternative one would be definitely good.

andyross, 2018-08-28:
This got merged in 1.12 but never closed.

closed 2018-08-28 by andyross