zephyrproject-rtos/zephyr #6533: 1.12 Release Checklist

carlescufi, 2018-03-09:
Add items as required
--------
- [ ] Major enhancements:
  - [ ] LLVM Support: Issue #3689
  - [ ] AMP support: Issue #3065
--------
- [ ] Release Notes WIP
  - [ ] Create draft doc/release-notes-1.12.rst with feature summary from https://www.zephyrproject.org/developers/#releases-overview
  - [ ] Update doc/release-notes-1.12.rst details sections (relative to 1.11.0 release) as important issues are addressed (subsystem owners)
--------
- [ ] Release data
  - [ ] Update version in VERSION
  - [ ] Run sanitycheck with --all --release and checkin resulting file scripts/sanity_chk/sanity_last_release.csv
--------
- [ ] Finalize Release Notes
  - [ ] Add list of GitHub issues
  - [ ] Review doc/release-notes-1.12.rst overview summary and details sections (see Sign-off below)
  - [ ] Update doc/release-notes.rst (add reference to 1.12 notes)
  - [ ] Update doc/index.rst (add reference to doc/1.12.0)
  - [ ] Create release and add notes on https://github.com/zephyrproject-rtos/zephyr/releases
  - [ ] Final tagged release as zephyr-v1.12.0 (with release notes)
  - [ ] Update https://www.zephyrproject.org/developers/#releases-overview
  - [ ] Update https://www.zephyrproject.org/developers/#downloads
  - [ ] Update doc generation (in future should be automatic based on tag)
  - [ ] Email devel@lists.zephyrproject.org & announce@lists.zephyrproject.org lists about release
--------
- [ ]  Final Release Notes Sign-off
  - Subsystems
    - [ ] Networking - @jukkar 
    - [ ] Bluetooth - @jhedberg , @carlescufi 
  - Architectures
    - [ ] ARC - @ruuddw 
    - [ ] ARM - @galak, @MaureenHelm 
    - [ ] X86 - @nashif, @andrewboie 
    - [ ] Xtensa, RISCV32, NIOS-II - @nashif, @andrewboie 
--------
- [ ] After Release
  - [ ] Update patchlevel in Makefile (PATCHLEVEL = 99)
  - [ ] Marketing Blog for announcing release (https://www.zephyrproject.org/blog)
  - [ ] Create branch for 1.12.x releases, update github protections for branch
  - [ ] Create checklist for next release

carlescufi, 2018-03-09:
Additional info: https://github.com/zephyrproject-rtos/zephyr/wiki/Release-procedure

MaureenHelm, 2018-06-11:
Next release: #8320

closed 2018-07-10 by MaureenHelm