zephyrproject-rtos/zephyr #6534: coap-server: Canceling Observation not working

baw-serafin, 2018-03-09:
On release 1.10 cancelling observation does not work

reproduce
--------------
* Get latest release (1.10)
* compile coap_server example with  `cmake -DBOARD=qemu_x86 ../..
* run in in coap with instructions from http://docs.zephyrproject.org/1.10.0/subsystems/networking/qemu_setup.html?highlight=networking%20qemu
* observe /obs resource with client (I used copper)
* make a lazy cancel (reset on notification)

expected
------------
* Notification should stop

result
--------
* Notification still arrive all 5s

I also tried explicit cancel (get with observe=1) but no result either

rveerama1, 2018-04-24:
@baw-serafin can you please try with latest upstream and share wireshark logs?

closed 2018-05-22 by nashif

baw-serafin, 2018-05-22:
@rveerama1 Thx for the fix, sorry I forgot to provide wireshark. COAP is unfortunatly not my top priority at the moment

commit ac3c6ba9 (2018-05-22): samples: net: Fix coap-server sample

If client removes observer then it reply with RESET message for
further observer notifications. That means coap-server should stop
sending further notifications to the client.

Current sample unref the packet as soon as it finds pending message
and doesn't bother about message type.

Fixes: #6534