zephyrproject-rtos/zephyr #6545: ARM Cortex R Architecture support

timork, 2018-03-10:
Hi,

Cortex R popular architecture widely used in real-time sensitive applications in automotive, storage controllers (HDD and SSD), robotics, aerospace and other areas.
Since no really popular RTOS have support Cortex R arch it can widely extend Zephyr applications coverage.

SoC platforms with Cortex R CPUs available for mass markets:
Cypress Traveo 
TI TMS570L
TI Hercules RM

+ Endless number of closed infrastructure ASICs 

Thanks

salerio-gs, 2018-10-04:
I would add the Xilinx US+ MPSoC (as found on 96 Boards Ultra96), this will become popular I think.

galak, 2018-10-04:
> I would add the Xilinx US+ MPSoC (as found on 96 Boards Ultra96), this will become popular I think.

That would be awesome!

galak, 2018-10-31:
> I would add the Xilinx US+ MPSoC (as found on 96 Boards Ultra96), this will become popular I think.

@salerio-gs is this something you are looking at or just saying would be nice?

PR #9316 (2019-08-09): Initial support for ARM Cortex-R
There have been a couple of requests for the addition of ARM Cortex-R support, e.g. https://github.com/zephyrproject-rtos/zephyr/issues/6545 and https://github.com/zephyrproject-rtos/zephyr/issues/5540.

I have started work for Cortex-R4 support for a project my company is doing.  This PR represents work that gets the "shell" sample working on an R4.  Basic context switching and interrupts using an R4 connected to a GIC400 are working.

The work and testing were done on a custom SoC so the PR does not include the SoC code since it won't be very useful to anyone [... cut]
closed 2019-08-09 by ioannisg