zephyrproject-rtos/zephyr #6577: sam0: SPI CS released too early

henrikbrixandersen, 2018-03-12:
I was struggling to debug my SPI device driver on the Adafruit Feather M0 Basic Proto, which uses the SAM0 SPI controller driver found in `drivers/spi/spi_sam0.c`.

Turned out the issue was not related to my driver, but to a defect in the "fast" paths of the `spi_sam0.c` driver.

I have used a simple test, abbreviated below, which transmits a command over SPI, discards the first received byte, and reads the reply to the command in the second byte:

	u8_t rdsr[] = { 0x05 };
	u8_t sr[] = { 0 };
	struct spi_buf tx[] = {
		{
			.buf = rdsr,
			.len = sizeof(rdsr),
		},
	};
	struct spi_buf rx[] = {
		{
			.buf = NULL,
			.len = 1,
		},
		{
			.buf = sr,
			.len = sizeof(sr),
		},
	};

	spi_transceive(&context->spi_cfg, tx, ARRAY_SIZE(tx), rx, ARRAY_SIZE(rx));

With the above code, the GPIO CS line is release halfway through the transfer as seen in the following screenshot from my oscilloscope (CH1 = SCK, CH2 = CS):

[image]

If I comment out the "fast" paths of the `spi_sam0.c` driver I get the expected behaviour of the GPIO CS line (it is released after the full transfer is completed):

    diff --git a/drivers/spi/spi_sam0.c b/drivers/spi/spi_sam0.c
    index 0ea3d01..a92111a 100644
    --- a/drivers/spi/spi_sam0.c
    +++ b/drivers/spi/spi_sam0.c
    @@ -361,17 +361,21 @@ static int spi_sam0_transceive(struct spi_config *config,
             * casing is 4x faster than the spi_context() routines
             * and allows the transmit and receive to be interleaved.
             */
    +#if 0
            if (spi_sam0_is_regular(tx_bufs, tx_count, rx_bufs, rx_count)) {
                    spi_sam0_fast_transceive(config, tx_bufs, tx_count, rx_bufs,
                                             rx_count);
            } else {
    +#endif
                    spi_context_buffers_setup(&data->ctx, tx_bufs, tx_count,
                                              rx_bufs, rx_count, 1);
     
                    do {
                            spi_sam0_shift_master(regs, data);
                    } while (spi_sam0_transfer_ongoing(data));
    +#if 0
            }
    +#endif
 
[image]

henrikbrixandersen, 2018-03-12:
One comment in `spi_sam0.c` puzzled me:

> This driver special cases the common send only, receive
> only, and transmit then receive operations.	This special
> casing is 4x faster than the spi_context() routines
> and allows the transmit and receive to be interleaved.

Out of curiosity, how was the 4x speedup measured?

nzmichaelh, 2018-03-12:
Hi Henrik.  Could you try changing:

static void spi_sam0_finish(SercomSpi *regs)
{
	while (!regs->INTFLAG.bit.DRE) {

with

static void spi_sam0_finish(SercomSpi *regs)
{
	while (!regs->INTFLAG.bit.TXC) {

and see if that fixes the problem?

re: speedup, this was by a mix of timer based benchmarking and hooking my signal analyser up to the port.  The generic routines have enough overhead that the bus isn't fully utilised, where the 'fast' versions interleave TX and RX and reduce the overhead.

nzmichaelh, 2018-03-12:
Oh, and the benchmark code is in https://github.com/zephyrproject-rtos/zephyr/pull/5730

henrikbrixandersen, 2018-03-12:
@nzmichaelh Changing it to checking the TXC flag does fix the CS line behaviour, but the data read is always 0xff (should be 0x04 according to oscilloscope, which is also the behaviour when commenting out the "fast" paths").

henrikbrixandersen, 2018-03-13:
@nzmichaelh At what SPI bus speed and CPU frequency was the 4x speed-up measured? I still can not wrap my head around how the fast paths could be 4 times as fast as the generic routines...

nzmichaelh, 2018-03-13:
Could you give https://github.com/nzmichaelh/zephyr/commit/d9c848c0516d4783a9ec6a07f63042e72e5b7519 a try?  I've tested it with the following cases:

* (tx one byte, NULL) (rx one byte) - aka the bug you reported
* (tx one byte, rx one byte) (rx one byte)
* (tx one byte, rx one byte) (rx five bytes)

nzmichaelh, 2018-03-13:
re: speed, there's two aspects: the generic code has a lot of overhead and it also doesn't interleave TX and RX.  From your screenshot it looks like you're running at ~100 kHz so the difference isn't visible but it makes a bigger and bigger difference at higher bus rates as you can interleave the CPU time with the IO time.

For example, here's some screenshots from the tests today with a 8 MHz bus:

Generic path:
[image]

Fast path:

[image]

Note how the bus is fully utilised during the 5 byte RX phase and that the total transfer time drops from 16 us to 8 us.  This effect increases with larger transfers (like when talking to a SD card) and higher bus speeds.

henrikbrixandersen, 2018-03-13:
@nzmichaelh Sweet! I have tested nzmichaelh/zephyr@d9c848c locally. It looks good to me and it fixes my initial issue.

Thank you for the comprehensive explanation of the speed-up using the fast paths. Makes sense to me now. I was indeed running @ 100kHz as I had dialled the bus speed down to debug the issues with the SPI driver.

PR #6613 (2018-03-15): spi: sam0: fix CS and back-to-back transfers.
Fixes #6577.

* Wait for all ongoing transmits to complete before de-asserting CS.
* When doing a tx then rx, wait for the previous tx to complete before flushing the rx buffer.

Tested on the Arduino Zero against a Olimex MOD-NRF24L module.
commit bb9d5167 (2018-03-15): spi: sam0: fix CS and back-to-back transfers.

Fixes #6577.

Wait for all ongoing transmits to complete before de-asserting CS.

When doing a tx then rx, wait for the previous tx to complete before
flushing the rx buffer.

Tested on the Arduino Zero against a Olimex MOD-NRF24L module.
closed 2018-03-15 by galak