zephyrproject-rtos/zephyr #6584: scripts: extract_dts_inlcudes: activate correct bus master type

erwango, 2018-03-13:
Following #6401, it happens that a sensor could be used on different bus types: I2C or SPI.

This should be handled when generating CONFIG bits from extract_dts_includes.
So depending on bus type identified from dts, a config flag providing bus type should be defined and 
appropriate config flags should be generated.

erwango, 2018-03-13:
Depends on #6504

erwango, 2018-03-13:
@tbursztyka : You said in #6401: 

>  this is not possible yet to get a proper spi dts bindings. I'll come with these.

So once it is available, I'll work on current issue.

tbursztyka, 2018-03-13:
@erwango Actually I already started to work on SPI/dts on device side. That's how I got the limitation on the cs/gpio part (rest is straightforward), which requires extract_dts_includes improvements as well.

b0661, 2018-03-13:
@tbursztyka 
can´t you use the i2c-slave or spi-slave directive to attach a slave device to a specific type of bus?

tbursztyka, 2018-03-13:
@b0661 you mean what's described here dts/bindings/device_node.yaml.template ?

tbursztyka, 2018-03-13:
Sure that's ok, actually I already started to set the slave device properly where relevant.
The issue I have is when (and only when) the CS is delegated to a gpio line: and that I don't see how to do it properly. What I would like is something like: cs-gpio: <&gpioX N> (which would be optional of course)

And it would generate:
- one define for activating cs/gpio delegation on that slave
- one define for a gpio controller name
- one define for the gpio line number

Unless I missed it, I don't see how to do that with current dts/tools.

erwango, 2018-03-13:
@b0661 : Indeed

Also, we need to decide on what to do with yaml bindings. Today we define st,lsm6dsl.yaml is a i2c-device (includes i2c-device.yaml).
So either we define new yaml bindings and new compatibles: "st,lsm6dsl-spi" > deviate from Linux
But we should be able to detect that node parent is SPI or I2C type and define bus directly from that. In that case spi/i2c-device.yaml could not be included in sensor binding > it requires some different computation.

erwango, 2018-03-13:
@tbursztyka: I've some bits of GPIO generation in #5381 which use current tools. But it does not manage the "(and only when)"

b0661, 2018-03-14:
> one define for activating cs/gpio delegation on that slave

By the cs-gpio and the phandle a define for the  chip select gpio device can be generated. I do the same for pinctrl client devices. The define should enable the slave to get the gpio device driver name (I made that the label property of the device) and the pin. By this the slave driver can configure and access the appropriate GPIO pin.

> one define for a gpio controller name

The label device tree property should be used as driver name by any device driver. This way you know what define to generate following the phandle of cs-gpio (see above) to the gpio device. 

> one define for the gpio line number

Easy to generate already given in the cs-gpio directive.

> Unless I missed it, I don't see how to do that with current dts/tools

I also think you cant´t do that without extending extract_dts_includes.py. That is why I created the DTDirective class in https://github.com/zephyrproject-rtos/zephyr/pull/5799/commits/1a744ed2ef36e942d7db6e539a0c341a67a1276b. You could use that commit and create a new class for [x]-gpio directives that always have the format [x]-gpio = <&[phandle of gpio] [gpio pin number]>.  

The principle could be:

    gpio_controller_node_address = phandles[p]
    gpio_controller_label = get_node_label_string(gpio_controller_node_address)
    ...
    prop_def[[slave-prefix]_[x]_GPIO_NAME]] = gpio_controller_label
    prop_def[[slave-prefix]_[x]_GPIO_PIN]] = gpio_pin

b0661, 2018-03-14:
@erwango
I think for sensors that are multi bus capable (e.g. SPI & I2C) there should be two different device tree specifications, for the two "faces" of the device. There may be a single driver that can imlement both faces of the device. By that the user can decide what face is relevant by just activating the respective device.

erwango, 2018-03-14:
@b0661 
> I think for sensors that are multi bus capable (e.g. SPI & I2C) there should be two different device tree specifications, for the two "faces" of the device.

Hence, we need to define 2 compatibles, "st,lsm6dsl-spi" and "st,lsm6dsl-i2c", with matching yaml bindinds, and we're good. These will be treated as 2 different sensors (which otherwise happen to have some common source code). Fine with me, only little modifications are required. No impact in extract script.

erwango, 2018-11-22:
Already implemented on one driver LSM6DSL. 
Closing

closed 2018-11-22 by erwango