zephyrproject-rtos/zephyr #6585: kernel: re-delete event list node [bug]

Likailiu, 2018-03-13:
reproduce:
1、the thread A use k_poll to register EVENT A.when add EVENT A into the queue,the interrupt occured.
2、in interrupt context, give a sem using _impl_k_sem_give(), the sequence is _impl_k_sem_give-
    >do_sem_give->handle_poll_events->_handle_obj_poll_events->sys_dlist_get(delete the node),and 
      clear  the state(polling) of thread A .
3、the thread B get schedule,registering EVENT B into the queue.the thread B will be blocked.
4、Thread A is scheduled to return to where it was previously interrupted,Where to perform is_polling() 
      branch,which is taken. but the step 2 already clear the state and delete the node.then it will re-delete 
      the node(sys_dlist_remove).this will cause registered EVENT B to be deleted by mistake.

Version:
1.92, 1.10.0 ,1.11.0

expected:
in the clear_event_registration(),  add   if(event->poller == NULL) {return;}
doing so can avoid double deletion.
result:
thread B  never get scheduled anymore.

pfalcon, 2018-03-15:
@Likailiu : Can you please provide the (minimal) code to reproduce this issue?

Likailiu, 2018-03-21:
[kernel_bug.txt](https://github.com/zephyrproject-rtos/zephyr/files/1832469/kernel_bug.txt)
bugs encountered in our actual hardware environment.
thank you

nashif, 2018-06-04:
no answer from original reporter, lowering priority.

Likailiu, 2018-07-05:
""clear_event_registrations" as a "re-delete the event node bug", but I don't see where you think it was deleted the first time."
in line 21 of the "kernel_bug.txt",  sys_dlist_get(delete the node);
pay attention to: inside of sys_dlist_get would call sys_dlist_remove(node) to remove the node.that is first time to delete the node

andyross, 2018-07-05:
I'm still confused.  Can you please submit code to exercise the bug?  Feel free to stuff irq_offload() calls into kernel code to cause context switches at the right times.

Likailiu, 2018-07-08:
thread A(Preemptable) register EVENT A:
thread context:
k_poll(event A)
{
	irq_lock();
	register_event();/*add EVENT A to list*/
	irq_unlock();
	//now interrupr occur ,to interrupt context 
	
}

interrupt context:
_impl_k_sem_give()
{
	do_sem_give()
	{
		handle_poll_events()
		{
			_handle_obj_poll_events()
			{
				sys_dlist_get(delete the node);/*first delete the EVENT A node */
				clear the state(thread A);/*thread A not polling*/
			}
		}
	}
}

after interrupt the thread B(may be timeout) get Scheduling:
thread B(non-Preemptable) register EVENT B;
thread context:
k_poll(event B)
{
	register_event();/*add EVENT B to list*/
	_pend_current_thread();
	swap();
}

now get back to thread A context:
thread A(Preemptable) register EVENT A,;
thread context:
k_poll(event A)
{
	irq_lock();
	register_event();/*add event A to list*/
	irq_unlock();
	 /*here the above interrupt is happened,the context would get back */
	//after interrupt and thread B 
	//get back to thread A context 
	
	if (!is_polling()) /*the thread A already is cleared the polling state in the above interrupt context, the above 22 line*/
		clear_event_registrations();//re-delete the event node.bug!!!
	
	
}

galak, 2018-12-06:
@andyross can you look at this again.

andyross, 2019-02-26:
Just got to this for like the third time, and... I'm sorry, I still don't see it.  In the example code, you have two threads operating on two separate events, how is it even possible for them to be confused?

@Likailiu - I genuinely believe you've found a bug, but without a reproducer (i.e. **actual code we can download and run**) I just don't see what can be done here.  Just describing the design using English isn't working.

andyross, 2019-03-04:
Yeah, I'm going to close this.  The poll code is indeed subtle, so I don't deny the possibility of a bug.  But I can't see it.  Note also that recent changes have made a double-remove from a dlist into a reliable runtime error, so if this were happening in practice it would be visible.  Please reopen with example code to reproduce if needed.

closed 2019-03-04 by andyross