zephyrproject-rtos/zephyr #6593: Allow configuring the USB serial number string in runtime

IvanSanchez, 2018-03-13:
As of now, the serial number string descriptor of the USB subsystem is a constant defined at compilation time:

https://github.com/zephyrproject-rtos/zephyr/blob/924e8aad124651b9d41f1acb0df4e384e6025f74/subsys/usb/usb_descriptor.c#L620

However, there is a *very* strong use case to flash the same firmware image into more than one board, and have them report different serial numbers through USB. In particular, there is the use case of the nRF5 SoCs, which have a small region of read-only memory from which unique numbers are fetched to seed the bluetooth static address:

https://github.com/zephyrproject-rtos/zephyr/blob/5475de105b58ae78470e91e09d625898affe6498/subsys/bluetooth/controller/hci/hci.c#L1852

https://github.com/zephyrproject-rtos/zephyr/blob/5475de105b58ae78470e91e09d625898affe6498/subsys/bluetooth/controller/hci/hci.c#L1858

Using those values can provide unique serial numbers. Other boards might have different methods of creating unique serial numbers (e.g. by reading a section of flash memory, or querying a peripheral when starting up).

I am also aware that developers can specify their own complete USB device descriptors, as illustrated in the WebUSB sample, but even there the descriptor is declared as `static`.

---

Thus, I request a way of building USB device descriptors allowing the developer to specify a (utf16-encoded) string for the serial number while keeping the rest of the structure static.

IvanSanchez, 2018-03-13:
cc @carlescufi

jfischer-phytec-iot, 2018-03-13:
@IvanSanchez I started to revise usb descriptor (see #4661 and #4767), I will consider it there. I think it can be implemented by something like usb_descriptor_board_fixup, usb device stack have to check if it is available and call it on initialization.

commit 32e6d0ca (2018-07-16): usb: set SN string descriptor at runtime

implement a method that can set SN string descriptor at runtime.
but runtime SN and default SN configured in Kconfig must has the
same length.

Fixes: #6593
closed 2018-07-16 by carlescufi