zephyrproject-rtos/zephyr #6605: Question: Does WAMP protocol has any usecase for Zephyr

akansal1, 2018-03-14:
I am looking for some solution similar to one described in [crossbar.io](https://crossbar.io/), where I can communicate between different systems running on different devices Constrained or Linux in my case.

I was looking for **OneM2M** based solution, but **WAMP** implementation by crossbar.io looks relevant as well

Does anyone here has any idea to about the **WAMP** and it is usecase for implementation in Zephyr for constrained devices?

Thnx

jukkar, 2018-03-15:
I have not seen any discussion of WAMP http://wamp-proto.org/ or any use cases so far. We have websocket supported in zephyr so probably WAMP could be implemented on top of that.

akansal1, 2018-03-19:
Yes, I was not able to find much over it as well, hence wrote it here
Found another thesis, that discusses and compare, WAMP and other protocols

https://into-archive.aalto.fi/download/attachments/12324178/Huang_Fuguo_thesis_2.pdf

Thx

jukkar, 2019-05-22:
As we do not have websocket support atm -> closing this for now.

closed 2019-05-22 by jukkar