zephyrproject-rtos/zephyr #6623: It is needed to support Application's Kconfig tree

jli157, 2018-03-15:
Some applications need sophisticated configuration requirements, thus need to define Kconfig tree on its own to support that. 

Before cmake based building system was implemented, an application can define its Kconfig tree by the following way in the Makefile:
```
KBUILD_KCONFIG = $(PWD)/Kconfig
export KBUILD_KCONFIG
```
and in its Kconfig:
```
config PROJECT_BASE
    string
    option env="PROJECT_BASE"

source "Kconfig.zephyr"
source "$PROJECT_BASE/src/Kconfig"
```

However, this way is not well supported in the current cmake based building system. The existing example which is using the most similar way to define its Kconfig is `tests/drivers/spi/spi_loopback`. The project redefined the KCONFIG_ROOT by the command
```
set(KCONFIG_ROOT ${CMAKE_CURRENT_SOURCE_DIR}/Kconfig)
```
in its CMakeLists.txt. Thus, an application level Kconfig can be defined. However, the application can't implement hierarchy Kconfig tree like below. This is why that example defines all of its new macros in one Kconfig file, which is cumbersome. 

So, a hierarchy Kconfig tree at the application level should also be supported.

jli157, 2018-05-08:
The PR #7293 has resolved the issue thus it can be closed.

closed 2018-05-08 by jli157

SebastianBoe, 2018-05-09:
Thank you for reporting :)