zephyrproject-rtos/zephyr #6635: tests/net/websocket/test doesnt build on qemu_xtensa

galak, 2018-03-15:
./scripts/sanitycheck -p qemu_xtensa -s tests/net/websocket/test 

Fails because completion doesn't exist.  This seems like a Kconfig issue:

warning: CONSOLE_HANDLER (defined at /home/galak/git/zephyr/drivers/console/Kconfig:30) has unsatisfied direct dependencies (UART_CONSOLE && CONSOLE), but is currently being selected by the following symbols:
CONSOLE_SHELL (defined at /home/galak/git/zephyr/subsys/shell/Kconfig:11), with value y, direct dependencies "y" (value: y)

galak, 2018-03-15:
Is NET_SHELL actually needed by tests/net/websocket/test?

jukkar, 2018-03-16:
> @jukkar Is NET_SHELL actually needed by tests/net/websocket/test?

No, the CONFIG_NET_SHELL=y in various unit tests is fully optional and not needed for the actual test. If found in unit tests, then it is probably a left over when debugging the actual test.

PR #6641 (2018-03-16): tests: net: websocket: Fix buiding on qemu_xtensa
The net-shell must be turned off in order to build on xtensa.

Fixes #6635
commit 24da695a (2018-03-16): tests: net: websocket: Fix buiding on qemu_xtensa

The net-shell must be turned off in order to build on xtensa.

Fixes #6635
closed 2018-03-16 by galak