zephyrproject-rtos/zephyr #6643: usb: nrf52 returns empty configuration responses

IvanSanchez, 2018-03-16:
I'm trying to get `samples/subsys/cdc_acm` running on a nRF52840 board (after sucessfully double-checking that it works after #6608 on an Arduino 101 board). The issue is that I'm seeing some strange behaviour during enumeration of the device.

I'm using a Debian Linux box for these tests. I'm using Zephyr freshly checked out from `master` (my `HEAD` points at https://github.com/zephyrproject-rtos/zephyr/commit/4304b4c38b23d7d4a5d9bf258eb6705fcd5f373f, right after #6608 was merged on 2018-03-16).

This is how my wireshark looks like. To the left is the nRF52840, to the right is an Arduino 101. Both are running the same freshly compiled & flashed `cdc_acm` sample. In both cases I've highlighted the transfer which contains the configuration descriptor:

[image]

To the right I can see the Arduino 101 with the expected behaviour:

* host→board: request device descriptor
* board→host: reply with device descriptor
* host→board: request configuration descriptor (max 9 bytes, to learn `wTotalLength`)
* board→host: reply with first 9 bytes of the configuration descriptor

However, the behaviour in the nRF52 doesn't look completely right:

* host→board: request device descriptor
* board→host: reply with device descriptor
* host→board: request configuration descriptor (max 9 bytes, to learn `wTotalLength`)
* board→host: **empty response**
* host→board: **re-request** configuration descriptor ([the linux USB stack retries up to 3 times on purpose](https://github.com/torvalds/linux/blob/3032f8c504d2b15d58e4c96060a96b47e215573c/drivers/usb/core/message.c#L645))
* board→host: reply with first 9 bytes of the configuration descriptor

This happens also on all the string descriptor requests in a similar, consistent fashion.

Basic functionality works, but triggering a code path in the linux stack designed for "flakey" [sic] devices doesn't feel right.

I've also tested nRF52840 hardware with some firmware images built with the nRF5 SDK, and the behaviour is as expected, so I can discard a hardware problem.

@sundar-subramaniyan Do you have any idea why these empty responses might be happening? I don't have the low-level expertise to know what might be going on.

cc @carlescufi

carlescufi, 2018-03-16:
CC @pawelzadrozniak

sundar-subramaniyan, 2018-03-16:
@IvanSanchez Could you upload the wireshark captures somewhere? (For both nRF52840 and Arduino 101 boards). I took a capture in my machine running Ubuntu with cdc_acm sample compiled from Zephyr master (HEAD pointing to 7f28edc) and I could see all the 67 bytes of the DEVICE DESCRIPTOR sent to the host - see attached screenshot. Maybe I'm missing your point. Let me understand it a bit with your captures.
[image]

IvanSanchez, 2018-03-16:
@sundar-subramaniyan Sure, thing, there they are:

[zephyr-bug-6643-arduino101.pcapng.zip](https://github.com/zephyrproject-rtos/zephyr/files/1819571/zephyr-bug-6643-arduino101.pcapng.zip)
[zephyr-bug-6643-nrf52840.pcapng.zip](https://github.com/zephyrproject-rtos/zephyr/files/1819572/zephyr-bug-6643-nrf52840.pcapng.zip)

The captures include the hub enumeration and address request/response. The arduino 101 capture includes the bootloader response, and the zephyr cdc_acm sample response after 5 seconds, as expected.

sundar-subramaniyan, 2018-03-16:
@IvanSanchez Thanks for uploading.

In your case the descriptor GET takes 2 tries every time.
However, I don't see the enumeration hiccups with my setup. But I still see garbled characters when I type something to the console.

Here's the capture from my machine.
[zephyr-bug-6643-nrf52_ubuntu.pcapng.zip](https://github.com/zephyrproject-rtos/zephyr/files/1819647/zephyr-bug-6643-nrf52_ubuntu.pcapng.zip)

IvanSanchez, 2018-03-16:
More info. The bug is reproducible on 52840 boards marked `QIAAAA` (PCB revision 0.9) but **not** on ones marked `QIAABA` (PCB revision 0.10). I'll have a look at the changelogs.

IvanSanchez, 2018-03-16:
It seems that this bug is caused by an underlying hardware bug present in the "Engineering A" boards (marked `QIAAAA`) but that was [fixed for the "Engineering B" boards](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52840.EngB.errata%2Fdita%2Ferrata%2FnRF52840%2FEngineeringB%2Flatest%2Ferr_840_fixed.html&cp=2_0_1_1_2) (marked `QIAABA`). 

That hardware is considered deprecated, so I'm closing this bug.

closed 2018-03-16 by IvanSanchez

sundar-subramaniyan, 2018-03-16:
@IvanSanchez That sounds like a HW issue. But you mention that nRF5 SDK based firmware works fine. Wondering if it applies some workaround, if any.

I've got a board with this marking on the SoC: QIAABB 1743AA
The Segger label reads: 0.11.0, 2018.3