zephyrproject-rtos/zephyr #6644: Bluetooth: Add reason parameter to L2CAP Channel disconnected callback

MariuszSkamra, 2018-03-16:
PTS test L2CAP/LE/CFC/BV-01-C requires user to be informed about Command Reject reason (sent as response to LE Creadit Based Connection Request).

Currently, L2CAP connection related callbacks looks as follows:
```
struct bt_l2cap_chan_ops {
	void (*connected)(struct bt_l2cap_chan *chan);
	void (*disconnected)(struct bt_l2cap_chan *chan);
};
```
would be nice to have disconnection reason parameter, just like in bt_conn_cb disconnected callback.
```
struct bt_conn_cb {
	void (*connected)(struct bt_conn *conn, u8_t err);
	void (*disconnected)(struct bt_conn *conn, u8_t reason);
};
```

jhedberg, 2018-03-20:
Sounds like a reasonable request to me. @MariuszSkamra do you want to create a patch for this?

MariuszSkamra, 2018-03-20:
I'm looking at this now, I see it is not so trivial. L2CAP disconnected callback is common for i.a. Command Reject and LE Credit Based Connection Response. Those have conflicting error codes:
```
0x0000 Command not understood
0x0001 Signaling MTU exceeded
0x0002 Invalid CID in request
```
vs
```
0x0000 Connection successful
0x0001 Reserved for Future Use
0x0002 Connection refused – LE_PSM not supported
```
What comes to my mind is to use ECONNREFUSED from errno.h. But it is not what I wanted to achieve here, there will be not L2CAP error code still. 
So probably I will use bluetooth testing API for this purpose.

closed 2018-03-20 by MariuszSkamra