zephyrproject-rtos/zephyr #6646: usb: protocol field in descriptor for CDC ACM should default to zero

IvanSanchez, 2018-03-16:
The protocol field here at https://github.com/zephyrproject-rtos/zephyr/blob/4304b4c38b23d7d4a5d9bf258eb6705fcd5f373f/subsys/usb/usb_descriptor.c#L207

should default to 0x00 (meaning "no protocol") instead of 0x01 (meaning "I'm actively advertising that I support the AT V.250 protocol" AKA `V25TER`).

Although this field has been ignored by major OSs during the last decades, this creates a big deal of problems with [ModemManager sending unexpected AT commands when plugging in a dev board](https://bugs.freedesktop.org/show_bug.cgi?id=85007). This has been mitigated up to some point by having `ID_MM_DEVICE_IGNORE` udev rules all over the place, which is just a stopgap measure.

This is a simple change, but I don't want to send a pull request before more work in #4661 is done due to fear of merge conflicts. Also, some applications might want to advertise support for some AT command set or ethernet emulation or a vendor protocol ([see table here](https://cgit.freedesktop.org/ModemManager/ModemManager/commit/?id=c4ea37fa02aaa0860f067062cbd315e3540aacce)), which is related to #4661 (refactor usb descriptors) and #6593 (let developers specify some values when putting together the usb descriptors).

cc @finikorg @jfischer-phytec-iot

finikorg, 2018-03-28:
@IvanSanchez OK with the change proposed, are you going to send a PR?

IvanSanchez, 2018-04-03:
@finikorg I'd rather wait for #4661.

nashif, 2018-05-29:
@finikorg can you submit a PR to fix this, looks like #4661 will never happen!

finikorg, 2018-05-30:
@nashif done

PR #8016 (2018-05-30): usb: cdc_acm: Set bInterfaceProtocol to No Protocol (0)
Do not set protocol in interface descriptor. Although this field has
been ignored by major OSs during the last decades, this creates a big
deal of problems with ModemManager sending unexpected AT commands when
plugging in a dev board.

Fixes: #6646
commit 73009d05 (2018-05-30): usb: cdc_acm: Set bInterfaceProtocol to No Protocol (0)

Do not set protocol in interface descriptor. Although this field has
been ignored by major OSs during the last decades, this creates a big
deal of problems with ModemManager sending unexpected AT commands when
plugging in a dev board.

Fixes: #6646
closed 2018-05-30 by nashif