zephyrproject-rtos/zephyr #6648: Trusted Execution Framework: practical use-cases (high-level overview)

ioannisg, 2018-03-16:
Introduction
---
This issue intends to list the practical use-case within the Trusted Execution paradigm.
By having a high-level description of each relevant use-case (requirements & challenges), we can facilitate the design of the Trusted Execution framework as part of Zephyr Project.

Use-case description
---

High-level description only:

1. **Secure Bootloader** 

   * Two images are loaded on the MCU: **Secure** bootloader and **Non-Secure** Zephyr RTOS + Application
   * MCU boots in Secure mode, from Secure bootloader. Bootloader verifies the Non-secure firmware.
   * After security attributions are configured, the bootloader prepares to boot to the  Non-Secure image.
   * Most of MCU resources (memory, peripherals, IRQs etc.) are allocated to the Non-Secure Zephyr firmware. Certain resources are allocated to Secure bootloader (e.g. secure memory, secure peripherals)

2. **Secure storage**

   * One composite image, or two separate images are loaded on the MCU.
   * Zephyr RTOS and user application runs in non-secure mode
   * Application uses _secure libraries_ (stored in secure image or in secure areas of the composite image 
 and accessible by non-secure callable entry functions) to perform secure storage operations (e.g. store critical data, keys, certificates etc. in the secure regions)
    * non-secure callable region shall be minimized, to remove wasted memory for the interface between the two security domains.

d3zd3z, 2018-12-04:
> non-secure callable region shall be minimized, to remove wasted memory for the interface between the two security domains.

Ideally, I think the secure veneers could just be an 'sg' followed by a jump to the implementation on secure side. The seem to end up being full functions in the current implementation of tfm, though.

carlescufi, 2020-03-10:
No longer relevant

closed 2020-03-10 by carlescufi

nkaramolegos, 2020-11-05:
i would like to store secure a private key in the device flash (nrf 52840-dk). I am bit confused does zephyr supports something like this?