zephyrproject-rtos/zephyr #6651: sanity tries to compile things when it should not

jukkar, 2018-03-16:
In several PRs, I have specified minimum ram usage in testcase.yaml. For example in PR #6588 I have set the "min_ram: 64" for those tests that can fit in that, and then there are couple of tests that need few kb more so in those tests the min_ram is 68.

When sanity runs these tests I see this in shippable log https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/12318/3/tests

```
Sanitycheck / nucleo_l432kc:tests/net/traffic_class/test_tc_8 / nucleo_l432kc/tests/net/traffic_class/test_tc_8 
...
FAILED: : && ccache /opt/sdk/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-gcc    zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr/zephyr_prebuilt.elf  -T zephyr/linker.cmd -Wl,-Map=/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/sanity-out/nucleo_l432kc/tests/net/traffic_class/test_tc_8/zephyr/zephyr.map -u_OffsetAbsSyms -u_ConfigAbsSyms -e__start -Wl,--start-group -Wl,--whole-archive libapp.a zephyr/libzephyr.a zephyr/boards/boards/arm/nucleo_l432kc/libboards__arm__nucleo_l432kc.a zephyr/subsys/net/libsubsys__net.a zephyr/subsys/net/ip/libsubsys__net__ip.a zephyr/subsys/net/ip/l2/libsubsys__net__ip__l2.a zephyr/tests/ztest/libtests__ztest.a -Wl,--no-whole-archive zephyr/kernel/libkernel.a zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -Wl,--end-group -L"/opt/sdk/zephyr-sdk-0.9.2/sysroots/armv5-zephyr-eabi/usr/lib/arm-zephyr-eabi/6.2.0/armv7e-m" -L/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/sanity-out/nucleo_l432kc/tests/net/traffic_class/test_tc_8/zephyr -lgcc -Wl,--print-memory-usage -nostartfiles -nodefaultlibs -nostdlib -static -no-pie -Wl,--fatal-warnings -Wl,-X -Wl,-N -Wl,--gc-sections -Wl,--build-id=none && :
Memory region         Used Size  Region Size  %age Used
           FLASH:       43432 B       256 KB     16.57%
            SRAM:       69028 B        64 KB    105.33%
        IDT_LIST:         132 B         2 KB      6.45/opt/sdk/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/real-ld: zephyr/zephyr_prebuilt.elf section `noinit' will not fit in region `SRAM'
/opt/sdk/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/real-ld: section .intList VMA [0000000020010000,0000000020010083] overlaps section noinit VMA [0000000020003470,000000002001088f]
/opt/sdk/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/real-ld: region `SRAM' overflowed by 3492 bytes
collect2: error: ld returned 1 exit status
```

In this case sanity should not try to execute this specific test for that board as that will not fit in ram. For this test case (test_tc_8) I have specified "min_ram: 68" in testcase.yaml file.

galak, 2018-03-16:
Is this because boards/arm/nucleo_l432kc/nucleo_l432kc.yaml  doesn't have ram/flash params in it.  I think PR #6617 would address that.

nashif, 2018-04-03:
ram is now added to those boards

closed 2018-04-03 by nashif