zephyrproject-rtos/zephyr #6654: efm32wg_stk3800 bluetooth sample do not compile

deadpoolcode1, 2018-03-17:
when trying to compile central_hr example for board efm32wg_stk3800, I get the following error: 

drivers/bluetooth/hci/h4.c:462:30: error: ‘CONFIG_BT_UART_ON_DEV_NAME’ undeclared (first use in this function)
h4_dev = device_get_binding(CONFIG_BT_UART_ON_DEV_NAME);

chrta, 2018-03-20:
So something like this is missing in the board's dts file, right?

~~~
#ifdef CONFIG_BT
                zephyr,bt-uart = &uart3;
#endif
~~~

Which UART and pins should be used for this?
I will not be able to test any implementation, since i do not have any bt hardware.

warner83, 2018-05-08:
Same issue also with cc2650_sensortag
Any workaround?

a46554, 2018-05-09:
I add 

`#ifdef CONFIG_BT
        zephyr,bt-uart = &uart0;
#endif
`
in boards/arm/cc2650_sensortag/cc2650_sensortag.dts.
Build success but can't scan the BLE signal from my CC2640 board

chrta, 2018-05-11:
I think that should be a dedicated uart that is not used otherwise, like for the console, right?

a46554, 2018-05-14:
bt-uart is use to communication between BLE host and BLE controller. CC2650 is totally BLE SOC solution like nrf52832. So that bt-uart shouldn't exist when TI CC2650 BLE function is support.

redradist, 2018-10-13:
Hi guys,

I have the same issue as @warner83 on **cc2650_sensortag**
This hardware is not supported yet or something missed in config file ?

PingiliV, 2019-04-11:
Hi, Iam getting this issue while compiling samples/bluetooth/beacon for psoc6. Is this issue resolved?
Any workaround?

chrta, 2020-11-04:
Just to clarify: The board efm32wg_stk3800 does not have any bluetooth hardware.

Question from me? Is there anything generic that can/should be done here? Or should we close this issue?

chrta, 2022-03-06:
Closing this because there was no feedback for over a year.

closed 2022-03-06 by chrta