zephyrproject-rtos/zephyr #6699: [Coverity CID: 183033] Program hangs in /tests/posix/pthread_rwlock/src/posix_rwlock.c

Shilpashree3lc, 2018-03-19:
Static code scan issues seen in File: /tests/posix/pthread_rwlock/src/posix_rwlock.c
 Category: Program hangs
 Function: test_rw_lock
 Component: Tests
 CID: 183033
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

nniranjhana, 2018-06-04:
This is a false positive, since `rwlock` is not locked in the first place (these are a series of negative tests for the rwlock APIs, and they assert EINVAL for all of them, because rwlock is not yet initialized at this point).

nniranjhana, 2018-06-07:
@nashif can close this issue

lpereira, 2018-06-07:
Indeed this is a false positive. Marked as Intentional in Coverity.

closed 2018-06-07 by lpereira