zephyrproject-rtos/zephyr #6719: compilation problems with posix/unistd.h

eyuval, 2018-03-20:
In my project I am using CONFIG_PTHREAD_IPC configuration option and compiling for the qemu x86 board. I get the following compilation error
/include/posix/unistd.h:23:19: error: conflicting types for ‘sleep’
 static inline int sleep(unsigned int seconds)
/zephyr-sdk-0.9.2/sysroots/i586-zephyr-elf/usr/include/sys/unistd.h:199:17: note: previous declaration of ‘sleep’ was here
 unsigned _EXFUN(sleep, (unsigned int __seconds ));

The same problem happens with usleep. Generally I am not sure whether I am supposed to include <posix/unistd.h> or <unistd.h> or likewise <posix/pthread.h> vs <pthread.h>

youvedeep, 2018-03-23:
Hi @eyuval  

 Can you please comment on which project you are trying to compile ?  Is that part of Zephyr tree ?

eyuval, 2018-03-25:
I am trying to compile a project out-of-tree, i.e. not part of zephyr

youvedeep, 2018-03-25:
Does your project has internal functions like sleep() and usleep ?

eyuval, 2018-03-26:
No it hasn't.

youvedeep, 2018-03-26:
I am still not able to reproduce your issue. As there are many sample application which has CONFIG_PTHREAD_IPC enabled but still it is working fine in Zephyr tree. (please have a look at tests in dir test/posix directory). 

But I found a potential issue in return type of sleep. I have created a PR https://github.com/zephyrproject-rtos/zephyr/pull/6804 for this. Can you please make this change locally inside your build system and see if it passes.

youvedeep, 2018-03-27:
yes that is next point we need to see,  The functions are declared as static inline in unistd.h and it is later on including #include_next <unistdh> file. So looks like the functions in side posix/unistd.h (static ) are coming in conflict with unistd.h file.

we should use <posix/xxx.h> due to that only if CONFIG_POSIX_IPC is enabled posix/unistd. h is getting included first. 
In order to avoid this conflict we need to mode sleep and usleep into a .c file and inside unistd.h file hae only declaration. I just created PR on this. Can you try if this helps you - https://github.com/zephyrproject-rtos/zephyr/pull/6820

unfortunately i can not test this on my local system as I am not able to reproduce this issue. As of now i have included Tool chain specifc file in guards based on how this file is included in test component. I do not have much idea on this file.

ramakrishnapallala, 2018-05-16:
@eyuval were you able check this from your end. I believe this issue is same as #6984