zephyrproject-rtos/zephyr #6727: k_mem_pool crash with larger values of n_max

andrewboie, 2018-03-20:
Steps to reproduce:

At toplevel declare:
K_MEM_POOL_DEFINE(my_pool, 64, 1024, 61, 4);

Where n_max is 61.

Then somewhere in code:
```
	struct k_mem_block blk;

	mem = k_mem_pool_alloc(&my_pool, &blk, 1024, 0);
```

This results in a page fault on x86, on further inspection, the dlist structures in pool->levels are being corrupted:

***** CPU Page Fault (error code 0x00000000)
Supervisor thread read address 0x20000003

If this is not a bug, we need to clearly document what the upper bounds for various parameters to K_MEM_POOL_DEFINE are, and add assertions to enforce them.

andyross, 2018-03-21:
Yeah, there's nothing wrong or out of bounds with that code.  This should work, it's a real bug.  Will investigate.

linkmeyer, 2018-06-05:
@andyross , what's the status?

andyross, 2018-07-24:
Last I tried this, I couldn't reproduce it.  But it needs investigation.  Will be fixed or closed for 1.13.

PR #9594 (2018-08-25): mempool: Fix bit pointer state for N_MAX > 31
When a mempool is created with a large number of maximum-size blocks,
the logic for initializing max_inline_level (i.e. when to union the
bitmask with the pointer and when to use the pointer directly) was
wrong.  The default state was "zero", which implies that level 0
should be inlined, but that's wrong with >32 base blocks.
Additionally, the type was unsigned, making the "level zero is a
pointer" situation impossible to represent.

Fixes #6727
commit 71ca6530 (2018-08-25): mempool: Fix bit pointer state for N_MAX > 31

When a mempool is created with a large number of maximum-size blocks,
the logic for initializing max_inline_level (i.e. when to union the
bitmask with the pointer and when to use the pointer directly) was
wrong.  The default state was "zero", which implies that level 0
should be inlined, but that's wrong with >32 base blocks.
Additionally, the type was unsigned, making the "level zero is a
pointer" situation impossible to represent.

Fixes #6727
closed 2018-08-25 by nashif