zephyrproject-rtos/zephyr #6749: kconfig: The error message is misleading when values are out-of-range

SebastianBoe, 2018-03-22:
Hi,

the error message is misleading when values are out-of-range. E.g.

CONFIG_FOO=5

     config FOO
         range 0 3

Should trigger an error about the value being out-of-range, but instead it triggers a warning about dependencies.

warning: FOO was assigned the value "5" but got the value "0" -- check dependencies

ulfalizer, 2018-03-22:
Might be tricky to infer that it was specifically due to the `range` from `kconfig.py`. Maybe something like this would point people in the right direction though:

```python
msg = "warning: {} was assigned..."
if sym.ranges:
    msg += " and range constraints"
print(msg, ...)
```

The behavior of `range` is a bit wonky in general. `default 5` is clamped to `3`, while 5 as a user value causes the lower end of the range to be used. It's only done like that for compatibility with the C implementation. It would make more sense to clamp both to 3.

SebastianBoe, 2018-03-22:
> The behavior of range is a bit wonky in general. default 5 is clamped to 3, while 5 as a user value causes the lower end of the range to be used. It's only done like that for compatibility with the C implementation. It would make more sense to clamp both to 3.

I'd say it would make more sense to error-out than to do any clamping.

The docs don't mention anything about clamping, but instead state that the user must
stay within the range. Which implies to me that an error should be triggered.

> - numerical ranges: "range" <symbol> <symbol> ["if" <expr>]
>   This allows to limit the range of possible input values for int
>   and hex symbols. The user can only input a value which is larger than
>   or equal to the first symbol and smaller than or equal to the second
>   symbol.

SebastianBoe, 2018-03-22:
I am happy with any solution that makes the user realize he is out-of-range.

SebastianBoe, 2018-03-22:
```
msg = "warning: {} was assigned..."
if sym.ranges:
    msg += " and range constraints"
print(msg, ...)
```
This should work.

ulfalizer, 2018-03-22:
A warning from Kconfiglib might be nice at least, when it detects that the user value is out-of-range. Should check that it doesn't get super spammy for the Linux kernel configs though.

Should get back to finding an accountant now. Code is more fun than IRL...

ulfalizer, 2018-03-23:
I added a warning for it: https://github.com/ulfalizer/Kconfiglib/commit/7ba79cafa8e03af90dea0c4378e024d24a7d64b7

SebastianBoe, 2018-03-26:
@ulfalizer : Much better, I would prefer an error, but if there are unresolvable compatibility issues with Linux then a warning is good enough.

closed 2018-03-28 by carlescufi

SebastianBoe, 2018-06-14:
I was just now prevented from using an invalid configuration due to this erroring-out at Kconfig.

Great job!

PR #6848 (2018-03-28): kconfiglib: Update to 7245bad9ebb58
Update Kconfiglib to upstream revision 7245bad9ebb58 (+ local Zephyr
modifications), to get the following improvements/fixes in:

 - Print a warning if an int or hex symbol is assigned a value that lies
   outside an active 'range' constraint.

 - Turn 'FOO' into a link in 'select FOO' when generating the Kconfig
   reference documentation.

 - Parenthesize '&&' expressions within '||' expressions --
   (A && B) || (C && D) is more readable than A && B || C && D.

The final two fixes will only be visible once the fix for #5622 gets in.

Fixes #6749
Fixes #68 [... cut]
commit 4b358757 (2018-03-28): kconfiglib: Update to 7245bad9ebb58

Update Kconfiglib to upstream revision 7245bad9ebb58 (+ local Zephyr
modifications), to get the following improvements/fixes in:

 - Print a warning if an int or hex symbol is assigned a value that lies
   outside an active 'range' constraint.

 - Turn 'FOO' into a link in 'select FOO' when generating the Kconfig
   reference documentation.

 - Parenthesize '&&' expressions within '||' expressions --
   (A && B) || (C && D) is more readable than A && B || C && D.

The final two fixes will only be visible once the fix for #5622 gets in.

Fixes #6749
Fixes #6 [... cut]