zephyrproject-rtos/zephyr #6755: stm32: Compile error if 2-nd UART console enabled

webhive, 2018-03-22:
Tried to enable serial console on UART2 for STM32F103C8 chip

```
CONFIG_UART_STM32=y
CONFIG_UART_STM32_PORT_1=y
CONFIG_UART_STM32_PORT_2=y
```
Getting issue during compiling

```
In file included from ../../../../include/arch/arm/arch.h:20:0,
                 from ../../../../include/arch/cpu.h:15,
                 from ../../../../include/kernel.h:33,
                 from <skipped>/zephyr/drivers/serial/uart_stm32.c:19:
zephyr/include/generated/generated_dts_board.h:75:49: error: ‘ST_STM32_USART_40004400_BASE_ADDRESS’ undeclared here (not in a function)
 #define CONFIG_UART_STM32_PORT_2_BASE_ADDRESS   ST_STM32_USART_40004400_BASE_ADDRESS
                                                 ^
zephyr/include/generated/generated_dts_board.h:75:49: note: in definition of macro ‘CONFIG_UART_STM32_PORT_2_BASE_ADDRESS’
 #define CONFIG_UART_STM32_PORT_2_BASE_ADDRESS   ST_STM32_USART_40004400_BASE_ADDRESS
                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
<skipped>/zephyr/drivers/serial/uart_stm32.c:413:1: note: in expansion of macro ‘UART_DEVICE_INIT_STM32’
 UART_DEVICE_INIT_STM32(2, STM32_CLOCK_BUS_APB1, LL_APB1_GRP1_PERIPH_USART2)
 ^~~~~~~~~~~~~~~~~~~~~~
zephyr/include/generated/generated_dts_board.h:76:49: error: ‘ST_STM32_USART_40004400_CURRENT_SPEED’ undeclared here (not in a function)
 #define CONFIG_UART_STM32_PORT_2_BAUD_RATE      ST_STM32_USART_40004400_CURRENT_SPEED

and so on
```

So lot of undeclared variables used. If stay single ``CONFIG_UART_STM32_PORT_1=y`` everything work fine. Single ``CONFIG_UART_STM32_PORT_2=y`` broken too.

erwango, 2018-03-26:
Hi @webhive ,

So, obviously some thing is missing here.
Though, it's hard to say what since STM32F103C8 chip is not in tree yet and everything depends on
what you've done to enable it.
Is there a branch I can look into in order to have a check and help you?

Erwan

erwango, 2018-04-20:
Changing priority to low as issue reported on out of tree code.
@MaureenHelm : What about removing all priority when issue reported is on out of tree code?
I think this kind of activity is rather more about community support than related to code quality.

ydamigos, 2018-04-20:
@webhive 

You need also to enable UART2 in your board's dts file.
E.g.:
```
&usart2 {
	current-speed = <115200>;
	pinctrl-0 = <&usart2_pins_a>;
	pinctrl-names = "default";
};
```

erwango, 2018-05-02:
@webhive , is issue still active?

webhive, 2018-05-02:
I think you can close it. Yes - it is a matter of dts.

erwango, 2018-05-02:
ok, closing, thanks!

closed 2018-05-02 by erwango