zephyrproject-rtos/zephyr #6764: ARC EMSK dts enhancements

ruuddw, 2018-03-23:
- add SPI and GPIO support, and whatever other peripherals are missing
- consider refactoring em7d, em9d, em11d soc/board/dts structure to avoid duplications and support new ARC development boards
- avoid the need for board fixups

nashif, 2018-03-23:
@ruuddw Does it make sense to split the em_starterkit board definition into 3 distinct boards 

- em7d
- em9d
- em11d
?

right now we are overloading all of those configs into one definition and this is starting to be confusing.

ruuddw, 2018-03-23:
@nashif  yes, splitting it into multiple boards is something we consider. However, since this target is an FPGA based development board that supports multiple configurations that only differ in the flavor of the ARC core used, that would lead to some duplication. Current model is one board with 3 SoC versions (different FPGA bitfile configurations).

vonhust, 2018-04-28:
@ruuddw @nashif  , I suggest to put em7d, em11d and em9d into one folder, named, snsps_emsk. As @ruuddw  said,  these cores share all the peripherals, the only difference is core related definitions, e.g. , core features, some interrupt configurations. These core can be viewed as in the same SoC family or SoC series.  If we split these cores into 3 distinct boards. The duplication will be unavoidable and make it difficult to maintain.

What's more,  in this year, we plan to add the support of a new protying board like emsk, but more advanced, which has 9 or more configurations.

I will submit a PR about cleanup the soc folders of arc for emsk soon.  The goal is to remove unnecessary duplication and make codes reusable.

vonhust, 2018-05-14:
done in #7238

ruuddw, 2018-05-15:
PR #7238 was merged.

closed 2018-05-15 by ruuddw