zephyrproject-rtos/zephyr #6773: Publish doxygen-generated API pages

dbkinder, 2018-03-24:
We use doxygen to process header files to collect public API information.  Today we use the XML output to feed into the Sphinx/Breathe processing to create the published [API documentation](http://docs.zephyrproject.org/api/api.html).  Doxygen also produces an HTML version of the API documentation that we generate but don't publish.  Some developers have asked for alternative ways of presenting the API docs and cross-references (e.g., #6739).  Publishing the doxygen-generated API material is another.  We should first though, do some customization of the doxygen output to make it more presentable and improve the user experience: Zephyr theme, intro on the doxygen index page, and such.

dbkinder, 2019-02-27:
Doxygen-generated API docs are published at https://docs.zephyrproject.org/apidoc/latest/modules.html but could use some improvements on the mainpage...

marc-hb, 2019-06-26:
Plus breathe is not actively developed right now, see for instance https://github.com/michaeljones/breathe/issues/420

PR #17075 (2019-06-27): doc: fix doxygen INPUT path; brand html output
Clean up the INPUT paths for doxygen scanning:
- directory scans include sub-directories so no need
  to include them explicitly
- remove excluded (legacy) includes that no longer exist

Sync directory changes in the CMakeLists.txt (as noted in the comments)

Add content to the home page of the doxygen-generated HTML, and add a
Zephyr branding theme.

Fixes: #6773
commit b30b52ab (2019-06-27): doc: fix doxygen INPUT path; brand html output

Clean up the INPUT paths for doxygen scanning:

- directory scans include sub-directories so no need
  to include them explicitly
- remove excluded (legacy) includes that no longer exist

Sync directory changes in the CMakeLists.txt (as noted in the comments)

Add content to the home page of the doxygen-generated HTML, and add a
Zephyr branding theme.

Fixes: #6773
closed 2019-06-27 by nashif