zephyrproject-rtos/zephyr #6782: Create PDF version of the project documentation

dbkinder, 2018-03-24:
Sphinx documentation generation can create LaTeX output that can then be used to create a PDF file, but we've not exercised that path in our documentation generation system.  Figure out the configuration tweaking and additional tools (e.g., ``pdflatex``) needed to make it so.

We also need the ability to create PDF versions for a subset of the documentation tree, e.g., Kernel documentation only.

pfalcon, 2018-09-21:
@dbkinder: Are we ready to enable PDF generation/download for http://docs.zephyrproject.org/ yet? I could use some re-reading of the docs with a PDF on a plane, but ... bummer.

nashif, 2018-09-21:
@pfalcon you can already create PDFs. You can also generate html and read it on the plane, unless you want to read things on some special device.

nashif, 2018-09-21:
@dcpleung isnt this resolved already?

nashif, 2018-09-22:
> In terms of having PDFs on the website, the CI needs to install some additional packages.

some... to get text up I had to install thousands :)

carlescufi, 2018-10-02:
I will be pushing a PR soon with the ability to generate PDF on Windows.

dbkinder, 2018-12-07:
We can generate PDFs (manually) now for the whole document tree, so closing this issue. Issue  #10324 tracks doing a PDF generation with a release.  No intention to add PDF generation with the daily build or CI check. The need for section PDF generation (much harder problem) will be looked into further.

closed 2018-12-07 by dbkinder