zephyrproject-rtos/zephyr #6785: Fail to compile when OT l2 debug is enabled.

ekarlso, 2018-03-25:
Hi if I enable OT debug for the nrf52840 application with the config below I get a error when building:
``` CONFIG_CPLUSPLUS=y
 
 CONFIG_SYS_CLOCK_TICKS_PER_SEC=1000
 
 CONFIG_BT=n
 
 CONFIG_NETWORKING=y
 CONFIG_ENTROPY_GENERATOR=y
 CONFIG_ENTROPY_NRF5_RNG=y
 
 CONFIG_NET_PKT_RX_COUNT=50
 CONFIG_NET_PKT_TX_COUNT=50
 CONFIG_NET_BUF_RX_COUNT=50
 CONFIG_NET_BUF_TX_COUNT=50
 
 CONFIG_NET_TX_STACK_SIZE=2048
 CONFIG_NET_RX_STACK_SIZE=3072
 
 CONFIG_NET_IPV4=n
 CONFIG_NET_IPV6=y
 CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=10
 CONFIG_NET_IPV6_NBR_CACHE=n
 CONFIG_NET_IPV6_MLD=n
 
 CONFIG_REBOOT=y
 
 CONFIG_NET_LOG=y
 CONFIG_SYS_LOG_SHOW_COLOR=y
 CONFIG_PRINTK=y
 
 CONFIG_UART_INTERRUPT_DRIVEN=y
 
 CONFIG_FLASH=y
 CONFIG_SOC_FLASH_NRF5=y
 CONFIG_FLASH_PAGE_LAYOUT=y
 CONFIG_MPU_ALLOW_FLASH_WRITE=y
 
 CONFIG_NET_L2_OPENTHREAD=y
 CONFIG_OPENTHREAD_DEBUG=y
 CONFIG_OPENTHREAD_L2_DEBUG=y
 CONFIG_OPENTHREAD_L2_LOG_LEVEL_DEBUG=y
 CONFIG_OPENTHREAD_CHANNEL=13
 CONFIG_IEEE802154_NRF5=y
 
 CONFIG_NET_APP=y
 CONFIG_NET_APP_NEED_IPV6=y
 CONFIG_NET_APP_CLIENT=y
 CONFIG_NET_APP_SETTINGS=y
 CONFIG_NET_APP_MY_IPV6_ADDR="fdde:ad00:beef::1"
 CONFIG_NET_APP_PEER_IPV6_ADDR="fdde:ad00:beef::2"
```

`
[123/138] Building C object zephyr/subsys/net/ip/l2/openthread/CMakeFiles/subsys__net__ip__l2__openthread.dir/openthread_utils.c.obj
FAILED: ccache /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-gcc -DKERNEL -DNRF52840_XXAA -DRADIO_CCA_CORR_LIMIT=2 -DRADIO_CCA_CORR_THRESHOLD=45 -DRADIO_CCA_ED_THRESHOLD=45 -DRADIO_CCA_MODE=NRF_RADIO_CCA_MODE_ED -DRADIO_PENDING_EXTENDED_ADDRESSES=1 -DRADIO_PENDING_SHORT_ADDRESSES=1 -DRADIO_RX_BUFFERS=1 -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I/home/ekarlso/software/zephyr/subsys/net/ip/l2/openthread/. -I/home/ekarlso/software/zephyr/subsys/net/ip/l2/openthread/../.. -I/home/ekarlso/software/zephyr/kernel/include -I/home/ekarlso/software/zephyr/arch/arm/include -I/home/ekarlso/software/zephyr/arch/arm/soc/nordic_nrf5/nrf52 -I/home/ekarlso/software/zephyr/arch/arm/soc/nordic_nrf5/nrf52/include -I/home/ekarlso/software/zephyr/arch/arm/soc/nordic_nrf5/include -I/home/ekarlso/software/zephyr/boards/arm/nrf52840_pca10056 -I/home/ekarlso/software/zephyr/include -I/home/ekarlso/software/zephyr/include/drivers -Izephyr/include/generated -isystem /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/lib/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/include -isystem /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/lib/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/include-fixed -I/home/ekarlso/software/zephyr/lib/libc/minimal/include -I/home/ekarlso/software/zephyr/ext/hal/cmsis/Include -I/home/ekarlso/software/zephyr/ext/hal/nordic/drivers -I/home/ekarlso/software/zephyr/ext/hal/nordic/nrfx -I/home/ekarlso/software/zephyr/ext/hal/nordic/nrfx/drivers/include -I/home/ekarlso/software/zephyr/ext/hal/nordic/nrfx/hal -I/home/ekarlso/software/zephyr/ext/hal/nordic/nrfx/mdk -I/home/ekarlso/software/zephyr/ext/hal/nordic/. -isystem zephyr/ext_proj/Source/ot/include -Os -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -imacros /home/ekarlso/eclipse-workspace/nrf52840/build/zephyr/include/generated/autoconf.h -ffreestanding -Wno-main -std=c99 -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Werror=implicit-int -ffunction-sections -fdata-sections -mabi=aapcs --sysroot /opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr -mthumb -mcpu=cortex-m4 -MD -MT zephyr/subsys/net/ip/l2/openthread/CMakeFiles/subsys__net__ip__l2__openthread.dir/openthread_utils.c.obj -MF zephyr/subsys/net/ip/l2/openthread/CMakeFiles/subsys__net__ip__l2__openthread.dir/openthread_utils.c.obj.d -o zephyr/subsys/net/ip/l2/openthread/CMakeFiles/subsys__net__ip__l2__openthread.dir/openthread_utils.c.obj   -c /home/ekarlso/software/zephyr/subsys/net/ip/l2/openthread/openthread_utils.c
/home/ekarlso/software/zephyr/subsys/net/ip/l2/openthread/openthread_utils.c: In function ‘add_ipv6_addr_to_ot’:
/home/ekarlso/software/zephyr/subsys/net/ip/l2/openthread/openthread_utils.c:119:124: error: ‘laddr’ undeclared (first use in this function)
   NET_DBG("Added %s", net_addr_ntop(AF_INET6,
                                                                                                                            ^    
/home/ekarlso/software/zephyr/subsys/net/ip/l2/openthread/openthread_utils.c:119:124: note: each undeclared identifier is reported only once for each function it appears in
/home/ekarlso/software/zephyr/subsys/net/ip/l2/openthread/openthread_utils.c: In function ‘add_ipv6_maddr_to_ot’:
/home/ekarlso/software/zephyr/subsys/net/ip/l2/openthread/openthread_utils.c:147:134: error: ‘laddr’ undeclared (first use in this function)
   NET_DBG("Added multicast %s",
                                                                                                                                      ^    
[123/138] Building C object zephyr/subsys/net/lib/openthread/platform/CMakeFiles/openthread_platform.dir/misc.c.obj
ninja: build stopped: subcommand failed.

`

ekarlso, 2018-03-26:
cc @KaSroka

KaSroka, 2018-03-29:
This will be fixed by @rlubos. Looks like variable names has diverged somewhere

ekarlso, 2018-04-22:
Any news on this?

rlubos, 2018-04-23:
Hi,
sorry for late response. I was trying to reproduce the issue but it appears that it is already fixed by this commit: https://github.com/zephyrproject-rtos/zephyr/commit/03b24082ee25d197bd976236c1e7270d0cd67299

It came in with OT support for KW41Z which was merged just recently (#6598). If you could just confirm that it works for you on current master I think we could close the issue.

rlubos, 2018-08-01:
Closing obsolete issue.

closed 2018-08-01 by rlubos