zephyrproject-rtos/zephyr #6802: Using k_thread_user_mode_enter() cause kernel crash

vakulgarg, 2018-03-26:
I am using nxp frdm_k64f (has cortex M4 core). 
I tried running zephyr/tests/kernel/mem_protect/userspace. It passes successfully.

However if I introduce a printf() in function userspace/src/main.c: umode_enter_func() under the condition when is_user_context is true, zephyr kernel crashes.

***** USAGE FAULT *****
  Executing thread ID (thread): 0x200002ec
  Faulting instruction address:  0x61a0
  Attempt to execute undefined instruction Caught system error -- reason 0

Further decoding faulting instruction address 0x61a0 using 'addr2line' takes me to userspace/build/frdm_k64f/zephyr/priv_stacks_hash.gperf:32
The given line number is inside following function (at the location where variable map is being dereferenced to get priv_stack_addr).

u8_t *_k_priv_stack_find(void *obj)
{
    const struct _k_priv_stack_map *map =
        _k_priv_stack_map_lookup((const char *)obj, sizeof(void *));
    return map->priv_stack_addr;
}

nashif, 2018-03-26:
@agross-linaro @andrewboie  FYI

nashif, 2018-03-26:
@andrewboie @agross-linaro  FYI

andrewboie, 2018-03-26:
This is specifically a problem with how printf() calls into the kernel.
We end up in the UART console driver's console_out() function, and a pointer value that is not accessible to user mode gets dereferenced. We need to already be in supervisor mode by the time we get to this point.
@vakulgarg please meanwhile use printk() which doesn't have this problem.
I'll work up a patch to fix this issue. Thanks @agross-linaro for helping out.

andrewboie, 2018-03-26:
Set to medium prio now that we understand the issue better.

vakulgarg, 2018-03-27:
It is not enough to change printf() to printk(). I tried following test_main() as user mode entry point. Printk() prints the given string fine, but there is a crash subsequently. 

void test_main(void *a, void *b, void *c)
{
        printk("Inside test_main\n");
        while (1) {
        }
}

***** Stack Check Fail! *****
Current thread ID = 0x20002eec
Faulting instruction address = 0x296ec

The crash location is at zephyr/kernel/thread.c:183.
 _k_except_reason(_NANO_ERR_STACK_CHK_FAIL).

agross-linaro, 2018-03-27:
Can you try removing your build directory and try again.  Or did you try
that already?

vakulgarg, 2018-03-27:
Remove build/ directory and rebuilding didn't help.

andrewboie, 2018-03-27:
OK we definitely have two issues here. The other problem seems to be specific to the ARM port.

andrewboie, 2018-03-28:
Changing title to reflect what @vakulgarg is seeing on ARM.
The printf() syscall thing, I opened a separate issue #6846

agross-linaro, 2018-03-29:
The generated syscall is breaking the ARM AAPCS.  Additional padding is being put in at the time of the svc to pad out the stack.  This is breaking the expected storage of parameters.

K_SYSCALL_DECLARE2(K_SYSCALL__ZEPHYR_FPUTC, _zephyr_fputc, int, int, c, FILE *, stream);
    1088:       b410            push    {r4}
    108a:       4603            mov     r3, r0
    108c:       460a            mov     r2, r1
static inline int _arch_is_user_context(void)
{
        u32_t value;

        /* check for handler mode */
        __asm__ volatile("mrs %0, IPSR\n\t" : "=r"(value));
    108e:       f3ef 8405       mrs     r4, IPSR
        if (value) {
    1092:       b954            cbnz    r4, 10aa <_zephyr_fputc+0x22>
                return 0;
        }

        /* if not handler mode, return mode information */
        __asm__ volatile("mrs %0, CONTROL\n\t" : "=r"(value));
    1094:       f3ef 8414       mrs     r4, CONTROL
    1098:       07e4            lsls    r4, r4, #31
    109a:       d506            bpl.n   10aa <_zephyr_fputc+0x22>
        __asm__ volatile(
    109c:       2470            movs    r4, #112        ; 0x70
    109e:       b084            sub     sp, #16
    10a0:       9402            str     r4, [sp, #8]
    10a2:       df03            svc     3
    10a4:       b004            add     sp, #16
    10a6:       bc10            pop     {r4}
    10a8:       4770            bx      lr
    10aa:       4611            mov     r1, r2
    10ac:       4618            mov     r0, r3
    10ae:       bc10            pop     {r4}
    10b0:       f7ff bfde       b.w     1070 <_impl__zephyr_fputc>

000010b4 <fputc>:

MaureenHelm, 2018-04-13:
Any update on this issue?

agross-linaro, 2018-04-13:
I just picked this back up.  I've reworked the __svc argument passing to do something different.   I am doing testing/validation now.

agross-linaro, 2018-04-13:
Note: Without Andrew's syscall patches the printf will still fail.  However, it should fail when it tries to access resources while not being in supervisor mode.  This is as expected, as there is not currently a way to trap the console accesses by the printf.

agross-linaro, 2018-04-30:
I tried reproducing with no success.  I mimic'd the above code and could not reproduce using just the single user_mode_enter test case or all of them in place.  I did see the usual cmake/make bug that required me to remove the build directory and rebuild the system.

vakulgarg, 2018-05-07:
I tried after pulling the latest zephyr code today. The above code still crashes, but with a different error.

> 
***** Stack Check Fail! *****
Current thread ID = 0x20002394
Faulting instruction address = 0x164f8
> 

My prj.conf has CONFIG_STACK_SENTINEL=y. 
Turning it off resolves the crash.

agross-linaro, 2018-05-09:
Ah, that's useful to know.  Ok, let me try again with the CONFIG_STACK_SENTINEL set

andrewboie, 2018-05-09:
Why are you using stack sentinel and user mode at the same time??
User mode has its own stack overflow detection, there is no need for the sentinel.

agross-linaro, 2018-05-09:
We need to make them mutually exclusive.  If USERSPACE is set, STACK_SENTINEL is turned off.  That's my two cents at least.

commit 09a8810b (2018-04-16): arm: userspace: Rework system call arguments

This patch changes the ARM system calls to use registers for passing
or arguments.  This removes the possibility of stack issues when
callers do not adhere to the AAPCS.

Fixes #6802
PR #7464 (2018-05-10): subsys: debug: Fix stack sentinel dependencies
This patch restricts the stack sentinel to only be allowed if the
USERSPACE configuration option is not set.  The stack sentinel feature
is redundant if used in conjunction with the USERSPACE, due to the
protection mechanisms in place for stacks.

Fixes #6802
closed 2018-05-10 by andrewboie