zephyrproject-rtos/zephyr #6822: Failed to ninja flash NUCLEO_L073RZ on windows

twowinter, 2018-03-27:
hello, i am new to zephyr.

I use NUCLEO_L073RZ to test zephyr, but failed on the step "ninja flash", these are log:

```
D:\winter\resources\zephyr\zephyr\samples\basic\blinky\build>ninja flash
[1/65] Generating always_rebuild
Building for board nucleo_l073rz
[1/2] Flashing nucleo_l073rz
Error: [WinError 2] 系统找不到指定的文件。
(Re-run as "D:/winter/resources/zephyr/zephyr//scripts/support/zephyr_flash_debug.py --verbose openocd ..." or set CMAKE_VERBOSE_MAKEFILE for a stack trace.)
FAILED: zephyr/cmake/flash/CMakeFiles/flash
cmd.exe /C "cd /D D:\winter\resources\zephyr\zephyr\samples\basic\blinky\build && C:\ProgramData\chocolatey\lib\cmake.portable\tools\cmake-3.9.2-win32-x86\bin\cmake.exe -E env C:/Python36/python.exe D:/winter/resources/zephyr/zephyr//scripts/support/zephyr_flash_debug.py openocd flash --board-dir=D:/winter/resources/zephyr/zephyr/boards/arm/nucleo_l073rz --kernel-elf=D:/winter/resources/zephyr/zephyr/samples/basic/blinky/build/zephyr/zephyr.elf --kernel-hex=D:/winter/resources/zephyr/zephyr/samples/basic/blinky/build/zephyr/zephyr.hex --kernel-bin=D:/winter/resources/zephyr/zephyr/samples/basic/blinky/build/zephyr/zephyr.bin --gdb=C:/gccarmemb/bin/arm-none-eabi-gdb.exe --openocd=OPENOCD-NOTFOUND --openocd-search= --cmd-load="flash write_image erase D:/winter/resources/zephyr/zephyr/samples/basic/blinky/build/zephyr/zephyr.bin 0x8000000" --cmd-verify="verify_image D:/winter/resources/zephyr/zephyr/samples/basic/blinky/build/zephyr/zephyr.bin 0x8000000""
ninja: build stopped: subcommand failed.
```

twowinter, 2018-04-03:
Is it about Openocd install ? or stlink path? I didn't find these information in the tutorial .

twowinter, 2018-04-09:
Waiting for this issue

erwango, 2018-04-09:
hi @twowinter ,

I can't test flashing on windows, so I'm not sure how to solve the issue.
But, for flashing, you can just drag and drop bin file (zephyr.bin) in the USB drive which is mounted when you plug the nucleo board to your PC.

twowinter, 2018-04-12:
hi  @erwango  

thank you very much, the mbed drag and drop method is good, sample work well.

i will keep on check the issue.

DWagenk, 2018-05-02:
I'm trying to eliminate the need of a VM for my development flow and am running into the same problems when using `ninja flash` or `ninja debug`  on windows. I can live without the flash target, but the debug target proved to be really helpful in the past.

Zephyr docs [getting-started - win](http://docs.zephyrproject.org/getting_started/installation_win.htm) (10.) suggest installing pyOCD, but that seems to lack support for STM32 boards.

Your error (and mine too) show, that zephyr is trying to use openocd. When I Install openocd and add it to the path, I get rid of the `--openocd=OPENOCD-NOTFOUND` part of the error, but I haven't found out yet how to give the right input for `--openocd-search=` which should be `C:\path\to\openocd\share\openocd\scripts`

DWagenk, 2018-05-03:
@erwango i haven't found, where I can pass something in to zephyr (at cmake configure time or when running ’ninja debug‘) to set the openocd search path (I read somewhere in non-zephyr context , that on windows openocd always needs to be provided with the search path explicitly)

DWagenk, 2018-05-03:
Hi there,
so I found the right env variable to set in `zephyr\cmake\flash\CMakeLists.txt`: 
```
cmake -DBOARD=nucleo_l073rz -DOPENOCD_DEFAULT_PATH=C:\path\to\openocd\share\openocd\scripts C:\path\to\zephyr\samples\hello_world
```
remember to have openocd installed and added to your path

I downloaded openocd for windows from [here](http://gnutoolchains.com/arm-eabi/openocd/) unzipped it and copied the `bin`and `share` dirs to `C:\Program Files\OpenOCD\` and added  `C:\Program Files\OpenOCD\bin` to path)

I added it to my %userprofile%\zephyrrc.cmd, so it now looks like this:
```
set ZEPHYR_TOOLCHAIN_VARIANT=gccarmemb
set GCCARMEMB_TOOLCHAIN_PATH=c:\gccarmemb
set OPENOCD_DEFAULT_PATH="C:\Program Files\OpenOCD\share\openocd\scripts"
```

Some documentation on this would be nice!

erwango, 2018-05-03:
@DWagenk , thanks for providing this solution.
Let me change title for this issue and get it implemented(documented) in tree

erwango, 2018-05-03:
@dbkinder : Do we have some directions already on how to flash with openocd that could be enriched with above directions?

erwango, 2018-05-04:
Closing this issue, as it is also handled in #7325 (under different tags, but under control anyway)

closed 2018-05-04 by erwango