zephyrproject-rtos/zephyr #6823: Optimize rendering of Kconfig documentation

mbolivar, 2018-03-27:
Generating and rendering all of the Kconfig pages in Zephyr's sphinx documentation is currently about 75% of the docs generation time in a profiling step I did recently.

This issue tracks optimizing this.

Ideas:

- Generate a thin page for each option, with a custom Sphinx directive that generates all of the text (using kconfiglib behind the scenes). This would avoid having to go to the trouble to generate RST, then parse it again

- Generate one page with all the options. This would allow having just one instance of a Sphinx directive, would avoid dirtying the tree with all the kconfig .rst files, and would potentially speed up the parsing even more

An additional benefit to having a directive is that Python directives can note dependencies on the input files they use to do their work, which could potentially speed up incremental builds.

dbkinder, 2018-03-27:
Quite true.  I too have noticed that this part of the doc generation really slows down the Sphinx processing.  The python script creates over 3400 .rst files (one for each Kconfig option and an index.rst to reference them all).  We might first start with whether all the information currently displayed for each option is that useful before finding a faster way to create it.  For example, is just the symbol name, it's description, and where it's defined sufficient?  Is the cross-reference information used? Or would it be best to have folks refer to the Kconfig file for details and we should keep this index of configuration options terse?

dbkinder, 2018-03-27:
... and just read #6821 that proposed a shorter list of config information.  Good.

ulfalizer, 2018-03-27:
I haven't looked into this, but it might be worthwhile to check if parallel processing can be enabled for `sphinx-build` as well. Currently it can't, due to `breathe`:

```
$ sphinx-build -j8 -t development -b html -d _build/doctrees  -q . _build/html
WARNING: the breathe extension does not declare if it is safe for parallel reading, assuming it isn't - please ask the extension author to check and make it explicit
WARNING: doing serial read
```

I checked with `top` that you only get a single process, even with `-j8`.

mbolivar, 2018-03-27:
> While the python script takes only a few seconds, the sphinx process time is probably around 10 minutes+ to chew on the .rst files and create the (3400+) HTML output files.

Yeah, I think if we did this with a Sphinx extension that populates nodes within a small and predefined set of rst files, that then get rendered, we have a solid chance at making this a lot faster.

> I haven't looked into this, but it might be worthwhile to check if parallel processing can be enabled for sphinx-build as well. Currently it can't, due to breathe:

Here's the (stale) upstream issue: https://github.com/michaeljones/breathe/issues/141. In general, Breathe is on life support: https://github.com/michaeljones/breathe/issues/336. So I think this is not likely to be fixed soon.

@ulfalizer, I've got a little experience writing Sphinx extensions, but I don't know anything about kconfiglib's details. Is your latest round of patches up somewhere? I might be able to "extension-ize" those so we can avoid the round-trip through disk, at least if we commit to doing everything on a single .rst page.

ulfalizer, 2018-03-28:
@mbolivar 
Almost done -- just some cleanup left to do. Here's the [current WIP](https://dpaste.de/x8ue) in the meantime ([output](https://www.dropbox.com/s/a34tlk2ncyus8po/promptandtype.png?dl=0)), if you want to get an idea of how it'll work.

The basic scheme hasn't changed since the old script: There's a separate RST file for each symbol, and an `index.rst` file that links to them. Is it possible to have a single file generate multiple logical pages in RST/Sphinx?

I'm off to bed now. My sleeping hours are a bit odd at times. :)

mbolivar, 2018-03-29:
> Is it possible to have a single file generate multiple logical pages in RST/Sphinx?

I have looked, and haven't found a way. I still hope I'm missing something, but I wouldn't bet on it at this point.

dbkinder, 2018-04-16:
You can create intra-page links, if that's what you mean by a "logical pages", so the index.rst could link to an anchor within a larger page.  The index.rst page today references the option-specific data with a simple ``:option:`CONFIG_2ND_LEVEL_INTERRUPTS` `` role, that links to the corresponding anchor defined with ``.. option:: CONFIG_2ND_LEVEL_INTERRUPTS`` directive.  That anchor could be in a large .rst file with all the options (or broken up into a few files) created by the script, instead of putting each option into its own .rst file.

ulfalizer, 2018-04-17:
@dbkinder 
The current index page already managed to crash Firefox a few times on my machine, so I'm not sure it's a good idea to make it larger. :)

I guess you could split the symbols up with many per page, though it feels a bit arbitrary from an organizational perspective.

ulfalizer, 2018-05-21:
@mbolivar ^ did you see that btw? :)

mbolivar, 2018-05-22:
> @mbolivar ^ did you see that btw? :)
> 
> Edit: Oh wait, it's still open.

I hadn't seen it. I'm subscribed to the issue now so I'll be sure to notice updates.

I have a vague memory of reading that Zephyr's docs make assumptions about breathe that were invalidated in later versions, so updating may involve a couple of hiccups, but I'm thirsty to cut that 1:30 of re-build time down by as much as possible, so I'll try to make time to chase them down if I find any. If we can get it into the 15-20 second range I think we're really starting to get in business.

melvinvermeeren, 2018-06-02:
Hopefully future PRs and releases will be handled quicker than this. Indeed Breathe has been experiencing a shortage of active developers for quite a while.

Please comment how much this has actually improved the build time once integrated, I am quite curious. During basic testing I did for the issue I forgot to take note of the actual build time.

mbolivar, 2018-06-04:
> Please comment how much this has actually improved the build time once integrated, I am quite curious. During basic testing I did for the issue I forgot to take note of the actual build time.

Thanks!

@dbkinder -- I believe there was at one point an issue requiring fixed Breathe and Sphinx versions as specified in requirements.txt: https://github.com/zephyrproject-rtos/zephyr/blob/master/scripts/requirements.txt#L2

Are we able to update to the latest of those projects now to integrate parallel builds, or would that break some other dependency?

melvinvermeeren, 2018-06-04:
I think it is safe to upgrade Breathe to 4.9.1 (or latest). However, for -j I do believe that Sphinx has been getting improvements and fixes in the 1.7.x series. It is probably best to update both.

You may get additional "duplicate declaration" warnings, which is being discussed about upstream. All it means is that if you refer to a function in the doc somewhere you will be not be sure to which declaration it will jump. I do not remember issue #, but I believe the solution that will be used in aliasing the non-primary declarations so there will be one "real" declaration.

The same problem exists in earlier Sphinx, but there was simply no warning emitted I think.

mbolivar, 2019-05-15:
With @marc-hb 's work and sphinx+breathe updates, the build is fast enough for me, so I'm going to close this for now.

closed 2019-05-15 by mbolivar