zephyrproject-rtos/zephyr #6831: Update template docs with build example

dbkinder, 2018-03-27:
We've got a sphinx extension used to generate example build instructions.  Include use of the ``zephyr-app-commands`` extension in the sample templates found in the ``doc/templates`` folder, to help show how to use it (vs. using a ``code-block``)

nashif, 2018-05-02:
@mbolivar I thought we had docs for the commands, can you please point us to where you have documented this?

mbolivar, 2018-05-02:
@nashif it's in pydoc:

https://github.com/zephyrproject-rtos/zephyr/blob/master/doc/extensions/zephyr/application.py#L16

I'll move the content to something under doc/templates and provide examples for the main use cases.

nashif, 2018-05-02:
@mbolivar thanks

mbolivar, 2018-05-02:
Switched assignment to @dbkinder as discused in #7312

PR #7353 (2018-05-11): doc: add doc writing guides w/common usages
I've collected some of the common issues encountered with doc reviews
into a new contributing document, and included use of the
Zephyr-specific extension for generating code building examples.

Updated conf.py and created an external list of substitutions making it
easier to manage them without editing the sphinx conf file (and
documented this).

Tweaked the comments in the application.py extension python code to
render better in the generated doc that extracts these comments (keeps
the documentation in the python code too to ease maintenance when
up [... cut]
commit 486c5a54 (2018-05-11): doc: add doc writing guides w/common usages

I've collected some of the common issues encountered with doc reviews
into a new contributing document, and included use of the
Zephyr-specific extension for generating code building examples.

Updated conf.py and created an external list of substitutions making it
easier to manage them without editing the sphinx conf file (and
documented this).

Tweaked the comments in the application.py extension python code to
render better in the generated doc that extracts these comments (keeps
the documentation in the python code too to ease maintenance when
u [... cut]
closed 2018-05-11 by nashif