zephyrproject-rtos/zephyr #6833: Question: extended advertising feature support for HCI

xiaom-GitHub, 2018-03-28:
Notice that link layer code has already implemented the extended advertising feature, and shell cmd "advx" and "scanx" for testing. May I know any plan about the extended advertising feature support for HCI cmd between host and controller? also including extended scan feature(e.g. receive the extended adv report and pass it to host). Thanks!

carlescufi, 2018-03-28:
The Link Layer currently only supports a very small subset of the Advertising Extensions feature, if that is what you are referring to. @cvinayak is currently working on rearchitecting the Link Layer to support multiple advertisers and scanners, which will pave the way to support the full set of Advertising Extensions. There is no timeline yet, but we hope to send the first patches very soon.

xiaom-GitHub, 2018-03-29:
Thanks for this quick confirm. Yes, also notice some "TODOs" listed in the code now. Adv extensions feature give more opportunities to avoid the large payload fragments, currently I just adjust the Legacy Adv data mtu (31 -> 251), HCI cmd buffer size and radio pdu buffer size to allow nrf52840 transmitting the large packet, however, it's just kind of workaround. Expecting the full set of Adv extensions feature. Thanks!

closed 2018-03-29 by carlescufi