zephyrproject-rtos/zephyr #6858: Default board configuration guidelines

erwango, 2018-03-29:
Most of the boards supported in Zephyr tree could be configured in quite multiple ways,
either for pins configuration, hardware interfaces, instances of these interfaces and
then configuration of the drivers for each of these entities.
When porting a board and pushing it upstream, one will either copy configuration from
a similar board, either configure the board according to his constraints or interesting
new ideas.
Outcome is a great heterogeneity of configurations and discussions during reviews.
Besides, these various configurations don't make things easy when one try to enable
tests/samples running on multiple boards.

Here are some guidelines to shave away board configuration multiplicity for default board configurations:

a) When a connector (arduino, 96board, ....) is present, configure pins to fit this connector
b) Configure ips matching these pins (For instance configure SPI instance for arduino SPI)
c) Configure board valuable HW and connectors (Button, Led, Sensors, USB, BT chip, Wifi,
    Ethernet, VCP, )
d) Configure an output for console
e) Propose and configure a default network interface
f) Enable all GPIO ports
g) Provide device tree Zephyr aliases

a) to f) could be done as of now.
g) will require some modification in dts based generation

To Do:
- [ ] Document these guidelines in zephyr doc
- [ ] Get existing boards to follow guidelines (a) to f))
- [ ] Enable device tree aliases generation

nashif, 2018-03-29:
You probably want this documented here: doc/porting/board_porting.rst

erwango, 2018-03-29:
> You probably want this documented here: https://github.com/zephyrproject-rtos/zephyr/blob/master/doc/porting/board_porting.rst

Thanks for the pointer!

erwango, 2019-09-19:
This has been implemented and documented.
If some boards don't follow this, bugs should be raised, but there is no point to keep this issue open.

closed 2019-09-19 by erwango