zephyrproject-rtos/zephyr #6866: build: requirements: No module named yaml

zhaozhongchen, 2018-03-31:
I am installing zephyr to a new ubuntu server according to this guide: Development Environment Setup on Linux. I had install additional packages by using this command:
```
pip3 install --user -r scripts/requirements.txt
```

after that i try compile an example for frdm_k64f,
```
$ cmake -DBOARD=frdm_k64f ../..
-- Generating zephyr/include/generated/generated_dts_board.h
Traceback (most recent call last):
  File "/home/qianfan/zephyr/scripts/dts/extract_dts_includes.py", line 14, in <module>
    import yaml
ImportError: No module named 'yaml'
CMake Error at /home/qianfan/zephyr/cmake/dts.cmake:121 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/qianfan/zephyr/cmake/app/boilerplate.cmake:260 (include)
  CMakeLists.txt:2 (include)
```

the build system complain that can't find yaml. so i install python3-yaml using apt-get. after that cmake generated Makefile successful. Unfortunately make can't find 'elftools':
```
ImportError: No module named 'elftools'
zephyr/CMakeFiles/offsets_h.dir/build.make:60: recipe for target 'zephyr/include/generated/offsets.h' failed
```

may be we forget add 'yaml' and 'elftools' in scripts/requirements ?

nashif, 2018-04-02:
Your python installation is not working correctly, i.e. when you install with pip3 install --user it does install libraries in your home directory, the installation path seems not to be in your python path

nashif, 2018-05-05:
works here.

closed 2018-05-05 by nashif

SebastianBoe, 2018-06-13:
This error message might occur if the wrong requirements.txt file is used.

e.g.

`pip3 install --user -r scripts/py2-requirements.txt`

vs.

`pip3 install --user -r scripts/requirements.txt`

mplebanski, 2018-12-20:
For those using `virtualenv` and still having this, make sure that you have sourced your virtual environemtn when running `cmake -GNninja -DBOARD=your_board ..`, otherwise ninja will generate build scripts for your default system interpreter. If that already happened just remove the `build` directory and run cmake again with environment sourced. I consider is useful also for people hitting #1392.

avisconti, 2019-04-10:
I had same issue this week myself, as I was forced to completely re-install a linux workstation and to
go again through the process of setting up the zephyr toolchain.

Few comments that may help other people in the future:

What I needed to do was to call following command:

pip3 install --user -r scripts/requirements.txt
-- or (better for my case) --
python3 -m pip  install --user --upgrade -r scripts/requirements.txt

I had to solve following issues:
1. http_proxy issues: I had '/' character in the pwd and phyton was not able to parse it.
    So, I changed the proxy password to avoid such character.

2. I had to upgrade setuptools:
python3 -m pip  install --user --upgrade setuptools

3. I finally ran 
python3 -m pip  install --user --upgrade -r scripts/requirements.txt

StefanHri, 2020-02-10:
I have a similar issue. Even when I do everything as suggested by @avisconti I still get:
```
CMake Error at /home/stefan/workspaces/zephyrproject/zephyrproject/zephyr/cmake/zephyr_module.cmake:45 (message):
  Traceback (most recent call last):

    File "/home/stefan/workspaces/zephyrproject/zephyrproject/zephyr/scripts/zephyr_module.py", line 18, in <module>
      import yaml

  ImportError: No module named 'yaml'

Call Stack (most recent call first):
  /home/stefan/workspaces/zephyrproject/zephyrproject/zephyr/cmake/app/boilerplate.cmake:429 (include)
  CMakeLists.txt:7 (include)

-- Configuring incomplete, errors occurred!
```