zephyrproject-rtos/zephyr #6890: tests: kernel: arm_irq_vector_table: Usage fault on nrf52_pca10040

spoorthik, 2018-04-02:
When zephyr.git/tests/kernel/arm_irq_vector_table is run on nrf52_pca10040, usage fault occurs.

Sample log:
```
 ***** USAGE FAULT *****
   Executing thread ID (thread): 0x20000d9c
   Faulting instruction address:  0x0
   Illegal use of the EPSR
 Fatal fault in ISR! Spinning...
```

Also, I could see following build warning: 

`warning: NUM_IRQS (defined at /home/kspoorth/work/latest_zephyr/arch/arm/soc/arm/beetle/Kconfig.defconfig.series:16, /home/kspoorth/work/latest_zephyr/arch/arm/soc/arm/mps2/Kconfig.defconfig.mps2_an385:12, /home/kspoorth/work/latest_zephyr/arch/arm/soc/atmel_sam/sam3x/Kconfig.defconfig.series:23, /home/kspoorth/work/latest_zephyr/arch/arm/soc/atmel_sam/sam4s/Kconfig.defconfig.series:22, /home/kspoorth/work/latest_zephyr/arch/arm/soc/atmel_sam/same70/Kconfig.defconfig.series:29, /home/kspoorth/work/latest_zephyr/arch/arm/soc/atmel_sam0/samd21/Kconfig.defconfig.series:29, /home/kspoorth/work/latest_zephyr/arch/arm/soc/nordic_nrf/nrf51/Kconfig.defconfig.series:23, /home/kspoorth/work/latest_zephyr/arch/arm/soc/nordic_nrf/nrf52/Kconfig.defconfig.nrf52832_QFAA:15, /home/kspoorth/work/latest_zephyr/arch/arm/soc/nordic_nrf/nrf52/Kconfig.defconfig.nrf52840_QIAA:15, /home/kspoorth/work/latest_zephyr/arch/arm/soc/nxp_imx/rt/Kconfig.defconfig.series:13, /home/kspoorth/work/latest_zephyr/arch/arm/soc/nxp_kinetis/k6x/Kconfig.defconfig.series:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/nxp_kinetis/kl2x/Kconfig.defconfig.mkl25z4:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/nxp_kinetis/kwx/Kconfig.defconfig.mkw2xd512:26, /home/kspoorth/work/latest_zephyr/arch/arm/soc/nxp_kinetis/kwx/Kconfig.defconfig.mkw40z4:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/nxp_kinetis/kwx/Kconfig.defconfig.mkw41z4:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/nxp_lpc/lpc54xxx/Kconfig.defconfig.series:13, /home/kspoorth/work/latest_zephyr/arch/arm/soc/silabs_exx32/efm32wg/Kconfig.defconfig.series:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f0/Kconfig.defconfig.stm32f030x8:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f0/Kconfig.defconfig.stm32f051x8:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f0/Kconfig.defconfig.stm32f070xb:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f0/Kconfig.defconfig.stm32f072xb:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f0/Kconfig.defconfig.stm32f091xc:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f1/Kconfig.defconfig.stm32f103xx:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f1/Kconfig.defconfig.stm32f103xx:33, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f1/Kconfig.defconfig.stm32f107xc:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f3/Kconfig.defconfig.stm32f303xc:18, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f3/Kconfig.defconfig.stm32f334x8:18, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f3/Kconfig.defconfig.stm32f373xc:18, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f4/Kconfig.defconfig.stm32f401xe:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f4/Kconfig.defconfig.stm32f405xx:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f4/Kconfig.defconfig.stm32f407xx:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f4/Kconfig.defconfig.stm32f411xe:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f4/Kconfig.defconfig.stm32f412cg:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f4/Kconfig.defconfig.stm32f412zg:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f4/Kconfig.defconfig.stm32f413xx:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f4/Kconfig.defconfig.stm32f417xx:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f4/Kconfig.defconfig.stm32f429xx:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f4/Kconfig.defconfig.stm32f446xe:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32f4/Kconfig.defconfig.stm32f469xi:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32l0/Kconfig.defconfig.stm32l072xz:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32l0/Kconfig.defconfig.stm32l073xz:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32l4/Kconfig.defconfig.stm32l432xx:15, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32l4/Kconfig.defconfig.stm32l475xg:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32l4/Kconfig.defconfig.stm32l476xx:15, /home/kspoorth/work/latest_zephyr/arch/arm/soc/st_stm32/stm32l4/Kconfig.defconfig.stm32l496xx:15, /home/kspoorth/work/latest_zephyr/arch/arm/soc/ti_lm3s6965/Kconfig.defconfig:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/ti_simplelink/cc2650/Kconfig.defconfig.series:14, /home/kspoorth/work/latest_zephyr/arch/arm/soc/ti_simplelink/cc32xx/Kconfig.defconfig.cc3220sf:10, /home/kspoorth/work/latest_zephyr/arch/arm/soc/ti_simplelink/msp432p4xx/Kconfig.defconfig.msp432p401r:22) was assigned the value "3" but got the value "39" -- check dependencies`

ulfalizer, 2018-04-02:
Hello,

From a Kconfig perspective, the problem is that `NUM_IRQS` isn't user-assignable (can't get its value from a `.config` file -- `tests/kernel/arm_irq_vector_table/prj.conf` in this case), because it has no `prompt`. Having a prompt corresponds to being visible in the `make menuconfig` interface, if you've used that.

Since `NUM_IRQS` isn't user-assignable, the user value from `prj.conf` is ignored, and the symbol falls back on the `default 39` in `arch/arm/soc/nordic_nrf/nrf52/Kconfig.defconfig.nrf52832_QFAA` instead.

Is the number of IRQs fixed or dynamic?

Looking into `arch/arc/Kconfig`, that arch makes `NUM_IRQS` user-assignable by adding a prompt. Something similar would need to be done for `arm` in order to make it possible to assign a value from `prj.conf`.

nashif, 2018-04-02:
The NUM_IRQ issue is a separate issue that we need to track somewhere else. We need to have a single definition of NUM_IRQ defined in 1 place for all architecture and use it in a consistent way, possible move it to arch/Kconfig and make it user-assignable

ulfalizer, 2018-04-02:
Yeah, seems a lot of stuff in those `Kconfig.defconfig` files could be handled via `.config` files instead.

ulfalizer, 2018-04-02:
Here's what I suspect the thinking behind the design is:

 1. Boards have some fixed setting/symbol X.

  2. If a user picks a particular board, they shouldn't be able to change X in e.g. `menuconfig`, so X has no `prompt`.

  3. Since X has no prompt, it can't be assigned its board-specific value via a `.config` file.

  4. X still needs to get its board-specific value somehow though, and that's what those `Kconfig.defconfig` files are for. You could have a single symbol definition with all the `defaults` too, but it would get spammy.

It would be a good idea to have that documented I think. Otherwise, people are likely to put stuff in random places.

If I've understood the design right, then I guess the "proper" fix would be to give the symbol a prompt in one of those arch- or board-specific `Kconfig` files, if it should be user-configurable on that particular machine.

As a note, you can put a condition on a prompt too, like `prompt "foo" if X`. If `X` is `n`, the symbol won't be user-configurable ("visible", in Kconfig speak).

ioannisg, 2018-04-03:
I did some initial investigation on this. and it looks to me that we have two separate issues.
- one is the `NUM_IRQS` redefinition in prj.conf, which is discussed above
- the other one is the Usage Fault that is reported, which I discuss below

First, this is reproduced, easily. I am not so familiar with the ISR table generation, but it looks to me that this examples
- sets `CONFIG_GEN_ISR_TABLES=n`, which does not allow the script to automatically generate the ISR tables.
- generates a custom vector table: 
```
typedef void (*vth)(void); /* Vector Table Handler */
vth __irq_vector_table _irq_vector_table[CONFIG_NUM_IRQS] = {
	isr0, isr1, isr2
```
But this vector table is not complete, since the example, most likely, sets up peripherals with IRQs, such as CLOCK_CONTROL, UART, RTC etc. Once these IRQs fire, null-pointer access generates the illegal EPSR exception.

I wonder if this test runs properly on other ARM boards. 
Copying @carlescufi @cvinayak @galak to get more opinions on this.

MaureenHelm, 2018-04-03:
This test passes on frdm_k64f, mimxrt1050_evk, and lpxcpresso54114

nashif, 2018-04-03:
works on sam_e70_xplained

ioannisg, 2018-04-03:
@nashif @MaureenHelm thanks for your input. 

I investigated this further, and turns out my initial assumption was valid: the UsageFault in nRF52832 is because somewhere - during initialization - either the RTC (timer) interrupt, or some interrupt associated with the BLE controller **fires**.

If I do not enable BT on pca_10040 and disable RTC1 interrupts, the test passes also on nrf52832.

PR #6926 (2018-04-08): Fix Test kernel arm irq vector table
This PR fixes the UsageFault error in irq_vector_table test for nRF52x-based platforms, reported in #6890.

Fixes #6890
commit 316ffff6 (2018-04-08): tests: kernel: fix irq_vector_table test for nRF52X platforms

This commit disables Bluetooth and adds a customized vector
table for nRF52X-based platforms in the irq_vector_table kernel
test, in order to prevent UsageFault during system timer
interrupts, and, therefore, let the test execute properly.

This commit fixes #6890.
closed 2018-04-08 by nashif