zephyrproject-rtos/zephyr #6893: Question: Struct data is erased for no apparent reason

chtavares592, 2018-04-02:
I'm developing a http_client zephyr application where the server sends a json, that json is stored in a structure.
But when I'm going to make a new request to the server, the struct data is erased .. for no apparent reason.
Does anyone know how to solve it?

jukkar, 2018-04-03:
Can you elaborate the issue more, it is a bit difficult to guess what you are doing?
Do you have sample code that shows the issue?

chtavares592, 2018-04-03:
Sorry, I'll try to explain it better.
I perform a post to the server to get a json, after getting it I perform the function call "json_obj_parse" to store the data in the struct. I verify the values contained in this structure and are all stored correctly. However, in my application it is necessary to perform a report to the server to inform the next steps, when I call the function "http_client_send_req" the data contained in the struct is deleted. I have already done the passing of this struct by parameter to the functions of my application to observe which moment the clean occurs occurs in the struct and I saw that it was at the time of calling this function.

jukkar, 2018-04-03:
Ok, perhaps this is related to json api usage. You can see one example how to use the json api in this PR #5035 
Checkout the PR and see samples/net/rpl_border_router/src/http.c file for examples of json.

jukkar, 2018-05-04:
@chtavares592 Is this issue still valid?

nashif, 2018-05-27:
@jukkar  ping, please provide latest status

jukkar, 2018-05-28:
No progress in this front. We would need a way to reproduce this, currently the issue is seen in out-of-tree application. I am closing this for time being, this could be re-opened with a proper way to reproduce the issue.

closed 2018-05-28 by jukkar

jukkar, 2018-05-28:
@otavio sure, I can review that.