zephyrproject-rtos/zephyr #6896: too many static MPU regions on many ARM targets

andrewboie, 2018-04-02:
Many SoCs are configuring up to five static MPU regions at boot. With the implicit regions used for thread stack, privileged stack guard, and app data memory, there are no free regions left on systems with 8 MPU regions.

We need to reconsider what we are doing here:

- For SoCs which have 8 MPU regions, ensure we have a background mapping for global supervisor mode read/write/no-execute access to the entire address space, and delete the rest of the boot time regions except for the FLASH region which should have permissions set properly for XIP.
- For non XIP systems we might need another region granted execute access for program text
- The rest of the regions, and peripheral I/O, PPB, Bootloader regions can simply be removed.

Need to check if NXP MPU has any different constraints.
Need to decide if we want to omit the background mapping and have more static regions on systems with more available MPU regions (16+)
Need to determine if shareable/cacheable region settings introduce any complication.

pizi-nordic, 2018-04-18:
I have thought for a while about a bit more complex MPU management, as we would like to protect more and more regions while the devices remains constrained.

In my opinion we could try to have dynamic MPU management allowing for unlimited number of regions. In such case the MPU regions registered  in the hardware will be only a subset of all regions defined with all other areas marked as protected. During the fault, the software could replace cold (unused for a longer period of time) region with the new hot one in order to provide access. If access is not allowed the same fault handler will raise the error.

Such approach has of course performance penalty, however it could be reduced by protecting some entries from removal. For example the entry covering code area might be marked as permanent, so it will be never removed by the fault handler.

agross-linaro, 2018-04-18:
Yeah the uvisor has something like this where they fault in things as needed.  This definitely makes things more complex and the cost for swapping could be higher than expected.

Aside from the stack guard, if we only wanted to restrict access when in user mode, we could drop all of the regions except the stack guard and only program what is needed while executing in user mode.  We could still exceed the number of regions, but at least then the swapping would be isolated to any user mode thread that is using too many regions.

pizi-nordic, 2018-04-19:
The solution depends on the direction we would like to go. For full blown OS, dynamic MPU management gives more flexibility (especially if we would like to support things like virtual memory in the future). However you are right that there is a cost: increased complexity and code size as well as some performance penalty.

But if we would like to keep Zephyr "simple", then ideas presented in  https://github.com/zephyrproject-rtos/zephyr/pull/6827 seems to be better. We might go a step further and integrate userspace compartment with the scheduler. Then, on task switch, the MMU will be filed by entries allowing for access to current thread and shared data structures. There will be no runtime faults which have to be handled and the userspace separation will be fully automatic. On the other hand such approach increase memory usage as all data and code have to be properly grouped and aligned, creating empty places in the address space. Creating build system which will automatically do the grouping basing on the architecture and CPU capabilities also might be a challenge.

andrewboie, 2018-05-24:
> We might go a step further and integrate userspace compartment with the scheduler. Then, on task switch, the MMU will be filed by entries allowing for access to current thread and shared data structures. 

This already happens with the memory domains.

PR #7895 (2018-05-24): arm_mpu: reduce boot MPU regions for various soc
The PPB, Peripheral, and any other I/O regions have been
removed, the Shareable options are only important with
unlocked dual cores. Just use the background mapping,
which is RWX for supervisor and no access for user mode.

The flash region needs to be kept to indicate read-only
policy. The RAM regions need to be kept to disable execution.

Fixes #6896
commit 6f6acb46 (2018-05-24): arm_mpu: reduce boot MPU regions for various soc

The PPB, Peripheral, and any other I/O regions have been
removed, the Shareable options are only important with
unlocked dual cores. Just use the background mapping,
which is RWX for supervisor and no access for user mode.

The flash region needs to be kept to indicate read-only
policy. The RAM regions need to be kept to disable execution.

Fixes #6896
closed 2018-05-24 by nashif