zephyrproject-rtos/zephyr #6908: shippable: console limit error (Console size exceeds 16 MB limit)

b0661, 2018-04-03:
I get this error message about console limit error when testing PR #5799. No errors found in PR.

  console_limit_error
  Console size exceeds 16 MB limit. Truncated from here.

see https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/12993/2/console

Thx
Bobby

nashif, 2018-04-03:
looks like you have way to many build errors, something I can see for example:

```
kernel.queue/zephyr/drivers/pinctrl/pinctrl_stm32.c
In file included from /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/include/kernel.h:24:0,
                 from /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/include/device.h:11,
                 from /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/arch/arm/soc/st_stm32/stm32f0/soc.h:27,
                 from /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/sanity-out/nucleo_f030r8/tests/kernel/queue/kernel.queue/zephyr/drivers/pinctrl/pinctrl_stm32.c:28:
/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/sanity-out/nucleo_f030r8/tests/kernel/queue/kernel.queue/zephyr/drivers/pinctrl/pinctrl_stm32.c: In function ‘pinctrl_stm32_mux_set’:
/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/sanity-out/nucleo_f030r8/tests/kernel/queue/kernel.queue/zephyr/drivers/pinctrl/pinctrl_stm32.c:2403:13: error: implicit declaration of function ‘pinctrl_tmpl_is_device_function’ [-Werror=implicit-function-declaration]
  __ASSERT((!pinctrl_tmpl_is_device_function(dev, func)),
             ^
/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/include/misc/__assert.h:95:9: note: in definition of macro ‘__ASSERT’
   if (!(test)) {                                     \
         ^~~~
cc1: all warnings being treated as errors
`
```

It is advised in this case to run sanitycheck locally and see what is going on..

b0661, 2018-04-03:
Thank you. Will check.

nashif, 2018-04-03:
you can run with --build-only, so qemu is not invoked.

b0661, 2018-04-03:
Thank you, that is the option I was looking for.

b0661, 2018-04-03:
Programming error that only shows up if ASSERT ON.

closed 2018-04-03 by b0661