zephyrproject-rtos/zephyr #6909: Number of regions in arm_core_mpu_buffer_validate() can overflow

lpereira, 2018-04-03:
The number of regions is an 8-bit value.  At the beginning of the for loop, r_index is set ot _get_num_regions+1, which might go past 255.  If an overflowed r_index is passed to _is_enabled_region(), _is_in_region(), _is_user_accessible_region(), things might work as expected.

lpereira, 2018-04-04:
One has to note that in most implementations, _get_num_regions() will return a value much smaller than 255.  However, adding 1 to the value returned by this function to convert from 1-based to 0-based indexing seems to me that a region is skipped in the validation.

Also, the ARM MPU implementation and the NXP MPU implementation seems to loop through the regions in a different way (one goes forwards, and the other goes backwards).  This also might be an issue.

andrewboie, 2018-05-16:
Fixed by #7382

closed 2018-05-16 by andrewboie